PX4/PX4-Autopilot #11832: Pixhawk 2.1 cube heater not completely off

jzazbert, 2019-04-10:
**Describe the bug**
We discovered that the heater in the cube is not completely off in 1.8.2 firmware, and that the green ring LEDs also increase the temp in the cube. Disabling the LEDs and completely shutting off the heater reduced sensor temps by 15-25 deg C. This is the new issue requested by @dagar to make sure this is looked at. https://github.com/PX4/Firmware/issues/5963

We have discovered that the heater is not 100% off as of 1.8.2 firmware. PX4 sets the io pin to be OPEN_DRAIN, and then sets it high, which leaves the heater control pin floating. Because of interactions with the resistor network and the transistor that powers the heaters, the heat is on a small amount, that gets worse as temperature increases.

Changing the drive pin for the heater to PUSH_PULL and setting to high, we get the heater to turn completely off. Both that it was partially on and that this fix turned it all the way off were experimentally measured on 2.1 hardware.

We also disabled the green "RING" LEDs as they were putting 0.3W into the cube, and in our design are not visible anyway.

With lights off and the heater completely off, we've seen 20-25C drop in sensor temperatures inside the cube. We had to make this change because we were pushing sensors out of spec at only 45-50C ambient. We can now calibrate to almost 60C ambient without pushing sensors past absolute operating max, and still have a little headroom. The MPU9250s get hotter than anything else, presumably because they are self-heating somewhat, but they also have a 105C absolute max operating that we do not exceed while doing temp calibration.

**To Reproduce**
Steps to reproduce the behavior:
1. Run cube with temp calibration bits enabled with stock 1.8.2 until all temps stabilize. Check temps on each sensor
2. Modify firmware to disable Green LED_RING, ensure LED_BLUE() never turns on, and set px4io GPIO_LED1 to push_pull to force heater off
4. Run same temp run with cube as in number 1, note difference in temps on the sensors.

**Expected behavior**
With this change, we saw temps drop by up to 25C.

**Drone (please complete the following information):**
- This was tested with a pixhawk 2.1 cube

**Additional context**

davids5, 2019-04-10:
@jzazbert Please test https://github.com/PX4/Firmware/pull/11833

I addressed only the Heater issue.

There is no consideration for the boot failure of Police lights. Code that fails at the level should never be deployed. 

I did not address the light ring as it is HW specific. To disable it on just has to undefine GPIO_LED4

I imagine we could add a parameter to stop the breathing at some point if this is a must, but the FMU will have read the parameter and the API to the PX4IO extended to support configuring it.

LorenzMeier, 2019-04-11:
I'm closing this - please re-open for the ring separately.

closed 2019-04-11 by LorenzMeier