PX4/PX4-Autopilot #11842: vl53lxx is not work with other board except Crazyflie

bys1123, 2019-04-13:
**Describe the bug**
The driver only detect PX4_I2C_BUS_EXPANSION without other I2C buses.

bys1123, 2019-04-13:
Try to fix:
I modified rc.board_sensors, add new line `vl53lxx start`.
Also `#define VL53LXX_BUS_DEFAULT PX4_I2C_BUS_EXPANSION2` in vl53lxx's drive.

Plug my module to I2C A, Pixhawk4 will not boot up successfully.
Buzzer music is normal, FMU ACT will not blink, I/O B/E light up.

bys1123, 2019-04-15:
I deleted `vl53lxx start`
Boot success, I tried start this driver in Console, still not work.
The bus setting looks right, I got i2c read like this:
[image]

bys1123, 2019-04-17:
Tested on 1.8.2:
```
nsh> vl53lxx start
VL53LXX i2c::transfer returened -125
ERROR [vl53lxx] driver start failed
```

bys1123, 2019-04-17:
Tried `vl53lxx start -a` on master branch firmware. Still not work.
I2C A port has no response.
This also break my console connection.

mcsauder, 2019-04-29:
@bys1123 , could you post the console output when you manually start the driver with `vl53lxx start -a` ?

bys1123, 2019-04-29:
continue in #11932

closed 2019-04-29 by bys1123