PX4/PX4-Autopilot #11847: Poor EKF2 yaw performance without compass

mhkabir, 2019-04-14:
**Describe the bug**
When using EKF2 without magnetometer fusion enabled, there is significant yaw drift. I'm using an ADIS16497 IMU (orders of magnitude better than the onboard sensors) on a Pixhawk 4 for tests.

This has been previously reported independently in https://github.com/PX4/ecl/issues/567 with analysis and logs.

**To Reproduce**
Steps to reproduce the behavior:
1. Set EKF2_MAG_TYPE to "None" or "MC Custom"
2. Reboot
3. Look at heading estimate

I also played with the inhibit imu bias estimation flag, but it didn't seem to make much of a difference.

**Expected behavior**
Drift lower than the order of degrees per second.

**Log Files and Screenshots**
Logs from boot and screenshots showing Z gyro integral compared to heading estimate. All tests are done with vehicle static on desk:

1. EKF2_MAG_TYPE = None (https://logs.px4.io/plot_app?log=edb5aada-9f19-424b-958d-9f932517fa9f)

[image]

2. EKF2_MAG_TYPE = None, EKF2_AID_MASK has "inhibit imu bias estimation" (https://logs.px4.io/plot_app?log=074aab41-8e42-41a8-a83e-9bd0430743a0)

[image]

3. EKF2_MAG_TYPE = MC Custom (https://logs.px4.io/plot_app?log=53912bb1-dd88-4168-b7a0-988edd0afa1b)

[image]

4. EKF2_MAG_TYPE = MC Custom, EKF2_AID_MASK has "inhibit imu bias estimation" (https://logs.px4.io/plot_app?log=6b17e1d2-6d03-4c29-8c01-ec087c4faf9a)

[image]

mhkabir, 2019-04-19:
@priseborough Can you please have a look?

priseborough, 2019-04-23:
There are a few issues here:

Bias estimation inhibiting only works on the delta velocity bias states - the parameter description here https://github.com/PX4/Firmware/blob/master/src/modules/ekf2/ekf2_params.c#L622 is ambiguous so I will raise a change to make it clearer. SW modifications are required to disable gyro bias estimation. 

The problem with disabling any form of yaw observation in a static environment with default IMU process noise is that quaternion state covariances grow until the filter loses stability.

[image]

The EKF2_ABIAS_INIT, EKF2_ACC_NOISE, EKF2_ACC_B_NOISE, EKF2_GYR_NOISE and EKF2_GYR_B_NOISE parameters will need to be retuned to match IMU performance. Replay is the best method of achieving this.

The magnitude of initial quaternion variance is set by the initial_tilt_err parameter which is current set at 0.1 rad. I would reduce that for an application with a high quality IMU and static alignment.

mhkabir, 2019-04-24:
In that case, it is probably worth re-tuning the default EKF IMU parameters which haven't been touched in a long time even though the sensor pipeline has undergone significant changes. These should not have change the noise characteristics of the underlying data, but we may be missing something.

Even with a standard Pixhawk IMU without temperature compensation, the drift is much higher than expected, which leads me to believe that there are suboptimal default parameters involved and that we have a potentially underperforming estimator **even in regular setups.** This likely goes unnoticed because of absolute magnetometer observations constraining the states.

priseborough, 2019-04-24:
We do not tune to optimise performance for regular setups, but to achieve reliable operation across a wide range of setups. The IMU parameters were set to be robust to bad IMU data and the filter is not tuned to operate without aiding for heading drift.

mhkabir, 2019-04-24:
Alright, thank you for the insight. Would you be able to post some notes on how to do replay tuning of the IMU parameters to get better results? I can help in getting them into the Devguide as well for visibility.

mhkabir, 2019-04-24:
High end IMUs also provide Allan Variance plots for the gyroscopes and accelerometers - I assume it should be possible get starter values for the EKF noise parameters from these?

[image]

[image]

zarathustr, 2019-04-24:
But with GPS for fixed wing MAVs, I think the yaw estimation would be satisfactory. This is due to the in-run stability characteristics of gyros.

mhkabir, 2019-05-04:
@priseborough Bump on documentation for re-tuning parameters.

mhkabir, 2019-08-02:
@priseborough bump on the above

bresch, 2020-05-13:
We need to make sure that this is fixed

moreba1, 2020-10-21:
Has this issue been resolved with the backup yaw estimator in px4 v1.11? (in FW and multirotor)

@dagar

bresch, 2020-10-21:
@moreba1 You could implement that as a new mag type. However, note that the yaw can start to drift if the drone isn't accelerating for a long period of time without magnetic aiding.

moreba1, 2020-10-21:
> 
> 
> @moreba1 You could implement that as a new mag type. However, note that the yaw can start to drift if the drone isn't accelerating for a long period of time without magnetic aiding.

If drones have acceleration during flight how much likely is it to drift?

bresch, 2020-10-28:
> If drone has acceleration during flight how much likely is it to drift?

It is has enough acceleration, the drift will be bounded and you should never loose local position.

tkazik, 2021-03-04:
This is still an issue, right? We experience some issues and might be affected by that too. Thx!

mamariomiamo, 2022-03-21:
We also experience yaw drift when using EKF2_MAG_TYPE = None
The zgyro starts to increase after taking off

bresch, 2022-08-15:
The ekf gyro bias estimate is now much more stable due to a stability improvement in the covariance prediction. The rest is probably tuning-related. Please re-open if this is still relevant and provide a recent log

closed 2022-08-15 by bresch