PX4/PX4-Autopilot #11848: Automatically set SENS_BARO_QNH based on GPS altitude

dagar, 2019-04-14:
At startup we could (optionally) automatically calculate an appropriate QNH such that the pressure altitude agrees with the GPS altitude.

https://github.com/PX4/Firmware/blob/d36b06f779d49d8dc7387bf2b4fa3aade999ada3/src/modules/sensors/voted_sensors_update.cpp#L863-L881

dagar, 2019-04-14:
Another interesting idea would be to query the weather data over mavlink, assuming that the GCS is internet connected.

dagar, 2019-04-14:
ADS-B weather data would be yet another option.

Antiheavy, 2019-07-14:
this issue is still worth keeping open for future work.

GigaFlopsis, 2024-10-30:
Hello everyone! What do you think about finally implementing this feature? It's been 5 years already, and auto-calibration is available even in INAV, but it's more relevant than ever now.

We can work on this. the code looks quite simple. However, I would add a number of restrictions:
1 Calibration should start a few seconds after power on.
2 Calibration should only happen under conditions of high GPS accuracy.
2 Recalibration should occur every 10 seconds while the drone is disarm.

garrettseepersad, 2025-01-29:
Worth considering