PX4/PX4-Autopilot #11851: Does CASA Outback DLL failsafe work?

raygamero, 2019-04-15:
I'm trying to make a system (based on 1.8.0) that has more advanced Data link loss behavior.
Instead of just RTL.  I would like it to go to a point and wait for a configurable period of time and THEN return to home and land.  
Is this possible with the CASA outback challenge failsafe?  Does this work on 1.8.0?

raygamero, 2019-05-02:
@dagar 
Can you comment on this.  I would like to test these different failsafe modes for DLL

julianoes, 2019-05-03:
@raygamero some things had been implemented for the UAV Outback Challenge in 2014. Since then, this code has probably not been used or tested. Therefore, all I can suggest, is that you look at the code and try it out (respectively debug it) in SITL simulation.

I would probably suggest to use master and not 1.8 because many improvements in general have been merged since.

hamishwillee, 2019-11-03:
@julianoes Do we just close this as Wontfix, or leave somehow open so that people can see it is something that could do with being stripped out.

julianoes, 2019-11-04:
The question has been answered (even if not conclusively) so let's close it.

closed 2019-11-04 by julianoes