PX4/PX4-Autopilot #11854: BMI055 IMU driver reports incorrect temperature

dmillard-TD, 2019-04-15:
The BMI055 IMU driver is reporting incorrect temperature results.
I've been discussing it on the forum here:
http://discuss.px4.io/t/pixhawk-4-temperature-compensation-failure/10226/10

All testing was performed on a Pixhawk 4 in an industrial environment chamber.

I've inspected the driver source and found that there is indeed a problem with the code that fills the report buffer with the temperature value.  The gyro driver always reports 0, and the accel driver is reporting some value between 23.0 and 23.9 no matter what the actual temperature is.

I've modified the driver code to read the temperature correctly using the slope and offset specified in the BMI055 datasheet.  I'll submit a PR after I've finished testing it.

dagar, 2019-04-16:
Thanks for reporting @dmillard-TD. I can quickly review and test once the PR is available.

dagar, 2019-04-16:
If possible I'd like to get this into v1.9.0.

dmillard-TD, 2019-04-16:
PR #11867 submitted.

dagar, 2019-04-17:
Fixed in https://github.com/PX4/Firmware/pull/11867

closed 2019-04-17 by dagar