PX4/PX4-Autopilot #11860: SD Card does not mount on v1.8.2 (pixhawk 4)

echoGee, 2019-04-16:
**Describe the bug**
The SD card does not mount on a Pixhawk 4. Unable to determine if this is a software issue or hardware. 

**To Reproduce**
Steps to reproduce the behavior:
1. Upload v1.8.2 stock firmware on pixhawk 4
2. power cycle autopilot with the stock 4GB SD card
3. Connect Qground control 
4. See if SD card is mounted using the _mount_ command on nsh. Only _/etc_ and _/proc_ is mounted. _/fs/microsd_ is not mounted
5. Trying to mount the SD card using _mount -t vfat /dev/mmcsd0 /mnt/fs_ crashes pixhawk 

**Expected behavior**
The SD card should be mounted. _mount_ command on NSH should return the _/fs/microsd_ as a mounted directory.
Alternatively, if this is a hardware problem with the pixhawk 4, there needs to be a way to identify and isolate this. Verified that the SD card works with a different pixhawk 4. 

**Log Files and Screenshots**
No log files are available since SD card can

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
Not mounted on a drone yet

**Alternative approaches tried to debug the issue:**

1. Uploaded master and beta branches to pixhawk 4. This however prevented the pixhawk from booting up.

mhkabir, 2019-04-16:
Can you try with a different SD card?

That said, attempting to mount should never crash the flight controller.

echoGee, 2019-04-16:
I did try with a different card. Same effect.

julianoes, 2019-04-17:
> Uploaded master and beta branches to pixhawk 4. This however prevented the pixhawk from booting up.

Which version of master was this. Can you try again and paste the git hash here?

echoGee, 2019-04-17:
> > Uploaded master and beta branches to pixhawk 4. This however prevented the pixhawk from booting up.
> 
> Which version of master was this. Can you try again and paste the git hash here?

This one . 
commit c0a1f5b72f99ae0170f3cfde03ddbdb25a8c1c68
Author: baumanta <tanja@auterion.com>
Date:   Wed Apr 17 19:20:29 2019 +0200

    Collision prevention improvements (#11866)
    
    * match max vel in col-prev to regular pos ctrl
    * change warning criterium to avoid float error
    * disable push back from obstacles
    * use floats consistently

echoGee, 2019-04-22:
Hi, anyone able to take a look at this ?

julianoes, 2019-04-23:
How is the SD card formatted that you are trying to use?

echoGee, 2019-04-25:
It is fat32 formatted. I've tried other sd cards with fat32 formatting

julianoes, 2019-04-25:
Sounds like faulty hardware, not sure what to do about it. If you have a Dronecode debug probe (or something similar) you could get the startup output.
https://dev.px4.io/en/debug/system_console.html

echoGee, 2019-05-15:
@julianoes : We found a way to get the debug probe and attaching the files here. 

A snapshot of the comparison of the debug probe:
Left: Debug probe without SD card. Right: Debug probe with SD card
[image]

[DebugProbeWithSDCard.txt](https://github.com/PX4/Firmware/files/3184631/DebugProbeWithSDCard.txt)
[DebugProbeWithoutSDCard.txt](https://github.com/PX4/Firmware/files/3184632/DebugProbeWithoutSDCard.txt)

davids5, 2019-05-16:
@echoGee - I would try curren master. If it does not run, and does not boot the px4io, nor mounts the SD card, I would say it is a HW problem and you should contact Holybro.

echoGee, 2019-05-17:
@davids5 I tried with master. Same problem. I've tried contacting Holybro. Yet to hear from them.

davids5, 2019-05-17:
I have asked them to give you a hand on Slack. You should get a response from them here

davids5, 2019-05-17:
@echoGee 

please send email to productservice@holybro.com with invoice

closed 2019-05-17 by davids5

arthurfromholybro, 2019-05-20:
> @julianoes : We found a way to get the debug probe and attaching the files here.
>
> A snapshot of the comparison of the debug probe:
> Left: Debug probe without SD card. Right: Debug probe with SD card [...]

Please write to productservice@holybro.com, our tech team will help you. Thanks!