PX4/PX4-Autopilot #11863: I want change the angle command

Parkdaewon1, 2019-04-16:
In some cases, I would like to give a roll/pitch command inside the code. I did this, but it doesn't work. I'd like to know how to give a roll/pitch command.

	struct vehicle_attitude_s att;
	memset(&att, 0, sizeof(att));

	orb_advert_t att_pub = orb_advertise(ORB_ID(vehicle_attitude), &att);

	if ((pixel_X)>0) {
		att.q[0] +=10;
	}
	else{
		att.q[0] -=10;
	}

	if ((pixel_Y)>0) {
		att.q[1] -= 10;
	}
	else{
		att.q[1]+=10;
	}
	orb_publish(ORB_ID(vehicle_attitude), att_pub, &att);

Parkdaewon1, 2019-04-16:
My Firmware ver. is 1.8.2

huuthe, 2019-04-22:
have you found a solution to this problem? What is pixel_X ? I do not know to set the value of roll, pitch. Thanks u so much!

julianoes, 2019-04-23:
You are trying to set an attitude setpoint but you are publishing an attitude. This means that you are publishing an information which is published by the estimator already (at a rate of 250 Hz).

If you want to publish the attitude setpoint you need to publish the [attitude setpoint message](https://github.com/PX4/Firmware/blob/master/msg/vehicle_attitude_setpoint.msg) and you probably need to disable the [position controller](https://github.com/PX4/Firmware/tree/master/src/modules/mc_pos_control) which otherwise "overrides" what you published immediately.

However, best would probably be if you take a step back and explain what you are trying to achieve. Then it's probably easier and more effective to help and make a suggestion on what you could do.

huuthe, 2019-04-23:
I am new, I want to write a simple program to control my tricopter, when I start the application, the motor turns with roll angles of 15 degrees, 0 degree pitch angle and my đrone at a height of about 1 m. Thanks you

julianoes, 2019-04-23:
I would suggest to use the Dronecode SDK and send offboard commands, you can create your binary and run it on a ground computer and command the drone using MAVLink:

https://sdk.dronecode.org/en/
https://sdk.dronecode.org/en/guide/offboard.html
https://github.com/Dronecode/DronecodeSDK/blob/develop/plugins/offboard/include/plugins/offboard/offboard.h

huuthe, 2019-04-23:
I just want to use my app to test tricopter on a test branch stand. Thanks you so much!

julianoes, 2019-04-23:
Ok, note that for the SDK you'll need a position estimate from GPS or flow currently, unless you make some modifications. Please follow up in https://github.com/Dronecode/DronecodeSDK for questions around the SDK.

closed 2019-04-23 by julianoes