PX4/PX4-Autopilot #11864: Auto Mission does not start from QGC with LPE 'fuse vision position' active

copterspace, 2019-04-16:
**Describe the bug**
When PX4 starts up with LPE = 'fuse vision position' active - it's not possible to start outdoor auto mission from QGroundControl, it reports 'Vehicle not ready'.
When I check 'fuse vision position' AFTER PX4 starts up - it works OK, starts mission, starts to fuse vision position once it becomes available.
Same issue was opened before, https://github.com/PX4/Firmware/issues/8018, I don't know why @EliaTarasov closed it.
 
**To Reproduce**
This bug I first met on my real drone, but it also can be reproduced in jmavsim:
1. Start jmavsim, connect QGC, set LPE_FUSION fuse vision position = 1,
2. reboot jmavsim
3. Try to Start mission - see message 'Vehicle not ready' in QGC
4. set LPE_FUSION fuse vision position = 0, reboot jmavsim
5. set LPE_FUSION fuse vision position = 1 after reboot
6. Auto mission starts OK

**Expected behavior**
Start auto mission no matter we have fuse vision position = 0 or 1.

**Log Files and Screenshots**
I attach simple mission I used in jmavsim, although it can be any mission you like
[testjmavsim.plan.txt](https://github.com/PX4/Firmware/files/3083890/testjmavsim.plan.txt)

**Additional context**
The solution is also described in https://github.com/PX4/Firmware/issues/8018.
I removed '&& !(_fusion.get() & FUSE_VIS_POS))' from line 60 of Firmware/src/modules/local_position_estimator/sensors/gps.cpp - it started to work OK in jmavsim.

EliaTarasov, 2019-04-18:
@copterspace Did you feed LPE on the step №1 by actual vision measurements?

copterspace, 2019-04-18:
No, because I do not have vision measurements in the moment of start/takeoff. 
My scenario is autonomous start of the drone (Zuza Visio) with down-facing camera from single Aruco marker (~165mm), perform mission and to land it right back onto this marker. When drone starts up/takes off - it does not see the marker yet.

EliaTarasov, 2019-04-18:
When you enable vision fusion and do not provide measurements then your LPE origin is not initialized. That's why you get error "vehicle not ready". In addition, you cannot fly auto mission with vision position enabled because this indirectly implies that you don't use GPS then (for LPE only, EKF2 allows it, but it fuses vision as odometry only).

copterspace, 2019-04-18:
Initialisation of LPE origin is task of onboard software, and it might be initialised only once I see markers (according to current local origin drone has in the moment I see marker first time). 
It can be initialised on particular waypoint, or I may not see visual markers at all during outdoor flight - in this case I can use only GPS. I can re-initialase visual origin several times during outdoor flight.

But `BlockLocalPositionEstimator::gpsInit` executes only once, when GPS initialised after startup.
After `_receivedGps = true;` (line 51) - line 60 simply blocks initialisation of global origin forever (until reboot). It will not initialise even if I later have correct vision measurements, because of line 49 `if (!_receivedGps) {` 
This makes LPE VPE unusable for outdoor GPS flights. But once we cancel vision position check during GPS init - it works OK, I tested. Actualy, before I found this code correction with line 60 - I tricked FCU - reboot with 'fuse vision position=0', then after power up and GPS init set 'fuse vision position=1' - then it flies OK

copterspace, 2019-07-16:
Dear colleagues, here's video example of my test flying mission: https://www.youtube.com/watch?v=RNG2Y_IHx6Y 
Without https://github.com/PX4/Firmware/pull/11865 it would not work.

TSC21, 2019-10-15:
Closed by https://github.com/PX4/Firmware/pull/11865.

closed 2019-10-15 by TSC21