PX4/PX4-Autopilot #11868: Critical service bulletin for Cubes Purchased between January 2019 to present

BazookaJoe1900, 2019-04-16:
As describe in this [link ](https://discuss.cubepilot.org/t/sb-0000002-critical-service-bulletin-for-cubes-purchased-between-january-2019-to-present/406)there is a critical issue with some cubes

I think its importent to follow the discution there, if it relevant, to identify such cubesand alert the user if he uses them.

I don't know more than what this link say....

rickyh777, 2019-04-18:
Does/can PX4 report a notice similar to the new functionality added to arducopter?

BazookaJoe1900, 2019-04-30:
No comments?
This seems like serious issue. I haven't hear about someone in PX4 that encounter crash because of that issue. but its seems like just matter of time.

mhkabir, 2019-04-30:
PX4 handles this safely if it occurs in-flight. It will also prevent arming and ground the vehicle (require recalibration) if a sensor is missing at bootup. See https://px4.slack.com/archives/C3GFT2QS3/p1555699096001500 and following messages.

BazookaJoe1900, 2019-05-07:
@mhkabir, so in that case, and if someone else can approve you can close the issue

julianoes, 2019-05-07:
I think @bkueng looked into this and can comment.

bkueng, 2019-05-07:
This is serious and I have looked into it, see https://github.com/PX4/Firmware/pull/11922 for reference.
Closing this, but feel free to reopen if you experience problems.

closed 2019-05-07 by bkueng