PX4/PX4-Autopilot #11869: Adding an Airspeed Sensor to a Generic Quadrotor x

NoahStorm, 2019-04-16:
Hello!

We are an Aerospace Engineering design team at the University of Minnesota trying to fly a quadcopter with an mRo Next-Gen MS5525 Airspeed sensor. We know that Qgroundcontrol does not allow an airspeed sensor to be added when using the Generic Quadrotor x airframe configuration by default. 

We are new to this, so I am not sure how to update the config file (if that is even the right approach) to allow for an airspeed sensor to be used. Has anyone had any luck putting an airspeed sensor on a quad? 

Thanks!

dagar, 2019-04-17:
Hi @NoahStorm, 

The only reason the airspeed sensors aren't started automatically on multicopters is because of the check here. https://github.com/PX4/Firmware/blob/3d36dbfba385a81e5bc7d814febd2ae991e2017b/ROMFS/px4fmu_common/init.d/rc.sensors#L34

We could probably remove that entirely if desired. QGroundControl might not show the option to calibrate, but you can still do it on the command line (`commander calibrate airsped`).

NoahStorm, 2019-04-17:
Thanks for your reply @dagar 

We tried the command you gave and we got the following error message:

[image]

Our air speed sensor calibrates just fine when we set the Pixhawk Mini to the fixed wing airframe configuration. How could we go about changing that if-statement in the sensors code?

julianoes, 2019-10-17:
@NoahStorm were you able to figure this out?

julianoes, 2020-02-12:
No answer, closing.

closed 2020-02-12 by julianoes