PX4/PX4-Autopilot #11870: Mag #0 fail when running SITL px4/gazebo

mhk4zemi, 2019-04-17:
Hi everybody

I am trying to set up a SITL with new px4, so I implemented my own customized model and tried to run a simulation while I am facing the following error:
`
ERROR [sensors] Mag #0 fail:  STALE!
ERROR [sensors] Sensor Mag #0 failed. Reconfiguring sensor priorities.
ERROR [sensors] Baro #0 fail:  STALE!
ERROR [sensors] Sensor Baro #0 failed. Reconfiguring sensor priorities.
`

here is the status of the sensors:
`
[sensors] gyro status:
INFO  [ecl/validation] validator: best: 0, prev best: 0, failsafe: NO (0 events)
INFO  [ecl/validation] sensor #0, prio: 100, state: OK
INFO  [ecl/validation] 	val:  -0.0178, lp:  -0.0007 mean dev:  -0.0000 RMS:   0.0112 conf:   1.0000
INFO  [ecl/validation] 	val:  -0.0346, lp:  -0.0070 mean dev:   0.0005 RMS:   0.0278 conf:   1.0000
INFO  [ecl/validation] 	val:   0.0012, lp:   0.0009 mean dev:   0.0001 RMS:   0.0041 conf:   1.0000
INFO  [sensors] accel status:
INFO  [ecl/validation] validator: best: 0, prev best: 0, failsafe: NO (0 events)
INFO  [ecl/validation] sensor #0, prio: 100, state: OK
INFO  [ecl/validation] 	val:   0.0557, lp:  -0.1779 mean dev:   0.0030 RMS:   0.4676 conf:   1.0000
INFO  [ecl/validation] 	val:  -0.2805, lp:   0.1872 mean dev:  -0.0038 RMS:   0.1723 conf:   1.0000
INFO  [ecl/validation] 	val:  -6.8190, lp:  -9.9094 mean dev:  -0.0180 RMS:   1.7720 conf:   1.0000
INFO  [ecl/validation] sensor #1, prio: 75, state: OK
INFO  [ecl/validation] 	val:  -0.4724, lp:  -0.1797 mean dev:   0.0083 RMS:   0.6641 conf:   1.0000
INFO  [ecl/validation] 	val:  -0.5386, lp:   0.1837 mean dev:  -0.0049 RMS:   0.3621 conf:   1.0000
INFO  [ecl/validation] 	val:  -2.9321, lp:  -9.8736 mean dev:  -0.0177 RMS:   3.3925 conf:   1.0000
INFO  [sensors] mag status:
INFO  [ecl/validation] validator: best: -1, prev best: 0, failsafe: YES (1 events)
INFO  [ecl/validation] sensor #0, prio: 1, state: STALE
INFO  [ecl/validation] 	val:   0.0000, lp:   0.0000 mean dev:   0.0000 RMS:   0.0000 conf:   0.0000
INFO  [ecl/validation] 	val:   0.0000, lp:   0.0000 mean dev:   0.0000 RMS:   0.0000 conf:   0.0000
INFO  [ecl/validation] 	val:   0.0000, lp:   0.0000 mean dev:   0.0000 RMS:   0.0000 conf:   0.0000
INFO  [sensors] baro status:
INFO  [ecl/validation] validator: best: -1, prev best: 0, failsafe: YES (1 events)
INFO  [ecl/validation] sensor #0, prio: 1, state: STALE
INFO  [ecl/validation] 	val:   0.0000, lp:   0.0000 mean dev:   0.0000 RMS:   0.0000 conf:   0.0000
INFO  [ecl/validation] 	val:  32.0000, lp:  32.0000 mean dev:   0.0000 RMS:   0.0000 conf:   0.0000
INFO  [ecl/validation] 	val:   0.0000, lp:   0.0000 mean dev:   0.0000 RMS:   0.0000 conf:   0.0000
INFO  [sensors] Temperature Compensation:
INFO  [sensors]  gyro: enabled: 0
INFO  [sensors]  accel: enabled: 0
INFO  [sensors]  baro: enabled: 0
INFO  [sensors] Airspeed status:
INFO  [ecl/validation] 	val:   0.0000, lp: 947654822309438423040.0000 mean dev: -37239354268624476047451095040.0000 RMS:      inf conf:   1.0000
INFO  [ecl/validation] 	val:  32.0000, lp:  31.9999 mean dev:   1.5679 RMS:   4.7712 conf:   1.0000
INFO  [ecl/validation] 	val:   0.0000, lp:   0.0000 mean dev:   0.0000 RMS:   0.0000 conf:   1.0000
`

julianoes, 2019-04-17:
Can you paste the whole output and put it between ``` please? That way it's easier to read, thanks.

mhk4zemi, 2019-04-17:
Thank you @julianoes , I have done it

julianoes, 2019-04-17:
Do these messages continue or only appear on boot up? If it's only on boot-up that might happen but it should go away once the simulation has established lockstep.

mhk4zemi, 2019-04-17:
The point is that these messages are only appearing once at the beginning of the simulation but perhaps because of this problem the position is not getting locked and I cannot take-off!
```
commander takeoff
ERROR [commander] rejecting takeoff, no position lock yet. Please retry..
Command 'commander' failed, returned 1.
```

And also do you know the reason why I am getting this error? Even I tried to reset the sensor manually 

```
sensor stop 
sensor start
```

but still, I got the same error for #Mag and #Baro.

julianoes, 2019-04-17:
Ok, and I assume you wait 30 seconds or so for GPS lock?

Can you do `uorb top` and paste the output please.

mhk4zemi, 2019-05-31:
thanks the problem solved. 
The problem was about setting the inertia matrix wrong.

closed 2019-05-31 by mhk4zemi