PX4/PX4-Autopilot #11873: FAiLURE: failed to hear from all subscribed simulation topics when trying launch gazebo simulation via mission_test.py

Dines97, 2019-04-17:
I was working on my project on PX4. I had to take a break for two weeks. Then I came back and tried to launch my project. I could not get. As a result, I just decided to reset the entire project and download the toolchain again. To make sure that I remember, I follow the tutorial  on https://dev.px4.io/en/simulation/ros_interface.html and run the python version of this example https://dev.px4.io/en/ros/mavros_offboard.html . Here are three consoles and what I started there.

First

`roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557"`

Second

`source ~/catkin_ws/devel/setup.bash    // (optional)
source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/posix_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/sitl_gazebo
roslaunch px4 posix_sitl.launch`

And third

`rosrun px4 mission_test.py MC_mission_box.plan`

And as far as I remember after the last drone should fly along a square path. But instead, after 60 seconds an error appears.

`rosrun px4 mission_test.py MC_mission_box.plan
No module named basemap

WARNING: mapping module requires basemap
    sudo apt-get install python-basemap
    or:
    conda install basemap

No module named basemap
[ROSUNIT] Outputting test results to /home/saytek/.ros/test_results/px4/rosunit-mavros_mission_test-MC_mission_box.plan.xml
[Testcase: test_mission] ... FAILURE!
FAiLURE: failed to hear from all subscribed simulation topics | topic ready flags: {'mission_wp': False, 'ext_state': False, 'global_pos': False, 'state': False, 'alt': False, 'local_pos': False, 'home_pos': False} | timeout(seconds): 60
  File "/usr/lib/python2.7/unittest/case.py", line 329, in run
    testMethod()
  File "/home/saytek/catkin_ws/src/Firmware/integrationtests/python_src/px4_it/mavros/mission_test.py", line 249, in test_mission
    self.wait_for_topics(60)
  File "/home/saytek/catkin_ws/src/Firmware/integrationtests/python_src/px4_it/mavros/mavros_test_common.py", line 248, in wait_for_topics
    format(self.sub_topics_ready, timeout)))
  File "/usr/lib/python2.7/unittest/case.py", line 422, in assertTrue
    raise self.failureException(msg)
--------------------------------------------------------------------------------
-------------------------------------------------------------
SUMMARY:
 * RESULT: FAIL
 * TESTS: 1
 * ERRORS: 0 []
 * FAILURES: 1 [test_mission]

`
What changed? What did I start doing wrong? Just my code is heavily tied to mission_test.py

archee8, 2019-04-22:
Hello! Try change second step on : 
In Firmware directory
`make px4_sitl_default gazebo`
`source ~/catkin_ws/devel/setup.bash`
`source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/px4_sitl_default`
`export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)`
`export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/sitl_gazebo`

julianoes, 2019-07-22:
Your error message tells you to do this:

```
No module named basemap

WARNING: mapping module requires basemap
sudo apt-get install python-basemap
or:
conda install basemap
No module named basemap
```

closed 2019-07-22 by julianoes