PX4/PX4-Autopilot #11875: 【bug？】Takeoff alt switch logic

xdwgood, 2019-04-19:
**Describe the bug**
I have seen some special cases, takeoff.cpp may produce the wrong choice

**To Reproduce**
When the following assumptions are true:
1. If `get_global_position()->alt`---->negative number eg: -11.0
2. If `navigator->get_takeoff_min_alt()`---->2.5
3. So https://github.com/PX4/Firmware/blob/master/src/modules/navigator/takeoff.cpp#L87 
`min_abs_altitude` = -8.5
4. If https://github.com/PX4/Firmware/blob/master/src/modules/navigator/takeoff.cpp#L95
`abs_altitude` = 3.0(User set alt)
5.https://github.com/PX4/Firmware/blob/master/src/modules/navigator/takeoff.cpp#L98
so `abs_altitude < min_abs_altitude` = false
6.https://github.com/PX4/Firmware/blob/master/src/modules/navigator/takeoff.cpp#L122
Resulting in a alt setting error, the vehicle will fly 11 + 4 = 15 meters

Is this assumption established?  @MaEtUgR

MaEtUgR, 2019-04-20:
Thanks for the report. Takeoff to the wrong altitude when flying close to sea level (which can result in negative reference altitude) looks familiar to me. You could have found the cause of the problem here. Let me check in SITL. I think we can just set the starting altitude of the simulator to below sea level to debug.

MaEtUgR, 2019-04-20:
I quickly tried SITL with the starting altitude set negative. Last number in https://github.com/PX4/jMAVSim/blob/79586deda5176c37de7804904f2553c8547edc4a/src/me/drton/jmavsim/Simulator.java#L76 for jmavsim. It worked as expected for a takeoff altitude which is still negative e.g. 2.5m above -11m AMSL and also with a positiove altitude e.g. 30m above -11m AMSL. Either I didn't understand it yet or it's tied to other conditions.

xdwgood, 2019-04-20:
If the param.7 set by QGC is relative height?

xdwgood, 2019-04-20:
could you check this vale ： get_global_position()->alt ? sitl log ？ thanks

MaEtUgR, 2019-04-20:
The param7 of `MAV_CMD_NAV_TAKEOFF` when sent as a `COMMAND_LONG` (which QGC currently does) is absolute altitude AMSL.

> could you check this vale ： get_global_position()->alt ? sitl log ？ thanks

On my way.

MaEtUgR, 2019-04-20:
I tested with debug output:
```
param7: -8.456331
alt: -11.008866
takeoff_min_alt: 2.500000
min_abs_altitude: -8.508866
home valid
abs_altitude: -8.456331
end_altitude: -8.456331
```
... didn't find the problem yet. Do you have a distance sensor to the ground? Just guessing here.

xdwgood, 2019-04-20:
I found the test on the Yuneec H520C.There are ultrasonic sensors, but they are not used by ekf outdoors.
I checked the yuneec native code and found that this logic is insufficient. I also checked px4 and found that we used the same error. So I raised this question in the px4 community.

xdwgood, 2019-04-20:
I found a difference: 
PX4 ： param7: -8.456331
my： param7: 3.0

So the question is, is this calculation done by qgc? Parameter 7 is not the takeoff height set by the customer?  
Did some conversions complete in qgc?

xdwgood, 2019-04-20:
I can report it using px4 on SITL @MaEtUgR

xdwgood, 2019-04-20:
[image]

I set the height to 3 meters, but it flew to 7 meters.

git diff:
```
+++ b/src/modules/commander/Commander.cpp
@@ -264,7 +264,7 @@ static bool send_vehicle_command(uint16_t cmd, float param1 = NAN, float param2
        vcmd.param4 = NAN;
        vcmd.param5 = (double)NAN;
        vcmd.param6 = (double)NAN;
-       vcmd.param7 = NAN;
+       vcmd.param7 = 3.0;
        vcmd.command = cmd;
        vcmd.target_system = status.system_id;
        vcmd.target_component = status.component_id;
```
`public static LatLonAlt DEFAULT_ORIGIN_POS = new LatLonAlt(47.397742, 8.545594, **-4**);`

MaEtUgR, 2019-04-21:
> So the question is, is this calculation done by qgc? Parameter 7 is not the takeoff height set by the customer?
Did some conversions complete in qgc?

As I wrote above (https://github.com/PX4/Firmware/issues/11875#issuecomment-485092539) if you send the command to takeoff `MAV_CMD_NAV_TAKEOFF` in a `COMMAND_LONG` message the altitude in `param7` has to be AMSL (Above Mean Sea Level)  and **not** above ground or home. The necessary addition of the vehicles current altitude and the desired takeoff altitude is calculated within QGC. In your original example -11 altitude, user wants 2.5 meter above, QGC adds them and sends -11+2.5=8.5 in `param7`, `abs_altitude = -8.5` which stands for absolute altitude (above sea level and not relative to the ground). It does what you expect it to do as long as the abolute altitude can be estimated.

I think it's desirable to support a relative takeoff altitude as parameter for simplicity but also for indoor/no GPS or baro applications. Maybe check https://mavlink.io/en/messages/common.html#MAV_CMD_NAV_TAKEOFF_LOCAL if you're looking for that.

xdwgood, 2019-04-21:
thanks, I close it......

closed 2019-04-21 by xdwgood