PX4/PX4-Autopilot #11876: How disable preflight check?

archee8, 2019-04-19:
Hello. When I launch drone from the ground, everything is fine(fig.1). But When I launch drone from the mobile robot, I get a preflight error. How disable preflight Check(fig.2)?
Fig.1.
[image]
Fig.2
[image]

[image]

AlexisTM, 2019-04-25:
The check that fails here is the following.

https://github.com/PX4/Firmware/blob/20adce50778f835b1787f17a168e0b1061bbb593/src/modules/ekf2/ekf2_main.cpp#L1776

You could change it to `_preflt_fail = false;` which will disable that specific Preflight check.

hamishwillee, 2019-07-25:
This is actually kind of interesting. This fails innovation checks, which are supposed to tell us if measurements vs IMU are mismatched with other sensors.

So it may be doing the right thing, but I wonder if this means it isn't possible to use PX4 from a moving target? @priseborough ?

closed 2021-01-13 by Jaeyoung-Lim