PX4/PX4-Autopilot #11884: Airspeed sensor not work on Pixhawk4 I2C A

bys1123, 2019-04-22:
**Describe the bug**
Current Airspeed sensor only opened on PX4_I2C_BUS_EXPANSION, but most of users will connect their airspeed sensor on I2C A (PX4_I2C_BUS_EXPANSION2) first. Makes users very confusing.

Code is here, but I didn't get how to change this driver, not similar with other I2C device.
https://github.com/PX4/Firmware/blob/master/src/lib/drivers/airspeed/airspeed.h#L48

LorenzMeier, 2019-04-24:
@bkueng Could you please look into this?

bkueng, 2019-04-24:
That should already be working, @DanielePettenuzzo fixed that, and I quickly tested it.
The airspeed sensor is only started for fixed-wing or vtols.
Are you testing master, and which airspeed sensor do you use?

bys1123, 2019-04-24:
@bkueng It's 4525DO on master branch for VTOL.

closed 2019-04-24 by bys1123