PX4/PX4-Autopilot #11885: Pyserial versioning check fails in px_uploader.py

Jwitty07, 2019-04-22:
When trying to upload the latest firmware, the upload fails with error 
`Error: pyserial not installed!
    (Install using: sudo pip install pyserial)
`
However, I already have this installed, both with python 2.7 and python 3.6.

I found that the version of pyserial that I have does not use the  `serial.__version__` syntax for checking the version, but rather uses `serial.VERSION`. Pyserial cannot be upgraded because part of my Linux system relies on this.

julianoes, 2019-04-23:
@Jwitty07 thanks for reporting and the information.

Could you please test if it works with the change here: #11887

Jwitty07, 2019-04-24:
The fix works. Thanks!

julianoes, 2019-04-24:
Cool, thanks for testing @Jwitty07.

PR #11887 (2019-04-24): px_uploader.py: fix version check
Presumably older versions of pyserial do not implement __version__ but only VERSION, so we need to check for that as well.

Fixes #11885.
closed 2019-04-24 by julianoes