PX4/PX4-Autopilot #11890: EKF2_NOAID_TOUT is not used and no_gps_timeout_max need a parameter

bigbellmercy, 2019-04-23:
The parameter EKF2_NOAID_TOUT is prepared in code but not used although it was intended to adjust dead-reckoning possible time. But currently if a variable 'no_gps_timeout_max' is increased it extends dead-reckoning possible time as several experiments. But it needs a code change because it is not linked to any parameter.

So EKF2_NOAID_TOUT is suggested to be linked to the 'no_gps_timeout' although some name may be changed to resolve these two problems by one time.

julianoes, 2019-04-24:
The param is defined here `_param_ekf2_noaid_tout` in:
https://github.com/PX4/Firmware/blob/0f346af226c8aa4823c08a24e522e7b218898a18/src/modules/ekf2/ekf2_main.cpp#L393-L394

and this param's actual internal variable reference is set in:
https://github.com/PX4/Firmware/blob/0f346af226c8aa4823c08a24e522e7b218898a18/src/modules/ekf2/ekf2_main.cpp#L587

And `_params->valid_timeout_max` is used in the ECL library:
https://github.com/PX4/ecl/blob/3f691891c48f4e87ae038a485a5f0de39039f0f7/EKF/ekf_helper.cpp#L1357

I think this is correct. Could it be that your problem is that you need to reboot after changing parameter?

bigbellmercy, 2019-04-25:
Thanks for good explanation and sorry for my misunderstanding.

closed 2019-04-25 by bigbellmercy

julianoes, 2019-04-26:
No worries, thanks!