PX4/PX4-Autopilot #11899: [MindPX] V1.9.0_beta3 not connecting to QGC on MAC computer

airmind, 2019-04-24:
**Describe the bug**
V1.9.0_beta3 not connecting to QGC on MAC computer

**To Reproduce**
Steps to reproduce the behavior:
1. Use a MAC computer (10.14.x/Mojave), launch QGC.
2. Connect MindPX to it.
3. It does not connect.

**Expected behavior**
Should auto connect.

**Log Files and Screenshots**
No mavlink message bytes are received on QGC side, though cu/tty device has been created under /dev.
[image]

**Additional context**
It can auto connect on Windows computer. And previous release like v1.8.2 works well on MACOS/QGC. 
So it looks mostly like the new Nuttx usb cdc/acm driver compatibility issue.

julianoes, 2019-07-24:
@airmind is this the same issue  as #12201?

airmind, 2019-07-24:
@julianoes 
Thanks for the touch.

I think it is a different one than #12201 
This issue is not related to any external power source, and this issue does not exist on a Windows system.

I did not test it against 1.9.2 yet, will let you know when I done.

julianoes, 2019-07-24:
@airmind ok. Do you have the startup console output when this fails?

airmind, 2019-07-25:
@julianoes 
I have tested against 1.9.2 and seems the issue still exists. Below is the startup output from a MindPX:

> WARN  [shutdown] Reboot NOW.
[boot] Fault Log info File No 4 Length 3177 flags:0x01 state:1
[boot] Fault Log is Armed
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
HW arch: AIRMIND_MINDPX_V2
FW git-hash: 106905871d374c125730fe675266d67c7f869469
FW version: Release 1.9.2 (17367807)
OS: NuttX
OS version: Release 7.28.0 (119275775)
OS git-hash: 423371c7d4012e725ac4ca51323a18df64e581b3
Build datetime: Jul  9 2019 13:48:21
Build uri: localhost
Toolchain: GNU GCC, 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
PX4GUID: 000100000000303038303435510c00370029
MCU: STM32F42x, rev. 3
nsh: mount: mount failed: No such device
INFO  [tune_control] Publishing standard tune 16
nsh: mkfatfs: mkfatfs failed: No such device
INFO  [tune_control] Publishing standard tune 2
INFO  [param] selected parameter default file /fs/mtd_params
Board defaults: /etc/init.d/rc.board_defaults
WARN  [dataman] Could not open data manager file /fs/microsd/dataman
ERROR [dataman] dataman start failed
rgbled on I2C bus 2 at 0x55 (bus: 100 KHz, max: 100 KHz)
nsh: rgbled_ncp5623c: command not found
nsh: rgbled_pwm: command not found
WARN  [blinkm] I2C init failed
WARN  [blinkm] init failed
ERROR [param] Parameter SYS_USE_IO not found
Board sensors: /etc/init.d/rc.board_sensors
WARN  [hmc5883] no device on bus 2 (type: 2)
WARN  [qmc5883] no device on bus 2 (type: 2)
WARN  [hmc5883] no device on bus 1 (type: 1)
WARN  [mpu6000] no device on bus #3 (SPI4)
INFO  [mpu9250] Bus probed: 3
WARN  [mpu9250] probe failed! 255
WARN  [mpu9250] no device on bus 3
LSM303D on SPI bus 4 at 2 (11000 KHz)
L3GD20 on SPI bus 4 at 1 (11000 KHz)
MS5611_SPI on SPI bus 4 at 3 (20000 KHz)
WARN  [bst] no devices found
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
WARN  [commander] Dangerously low battery! Shutting system down
WARN  [commander] System does not support shutdown
Starting Main GPS on /dev/ttyS3
Starting MAVLink on /dev/ttyS1
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
ERROR [mavlink] DM_KEY_MISSION_STATE lock failed
ERROR [mavlink] offboard mission init failed (-1)
INFO  [init] Mixer: /etc/mixers/quad_x.main.mix on /dev/pwm_output0
ERROR [param] Parameter CAM_CAP_FBACK not found
px4flow [331:100]
INFO  [px4flow] scanning I2C buses for device..
INFO  [logger] logger started (mode=all)
INFO  [logger] log root dir created: /fs/microsd/log
NuttShell (NSH)

julianoes, 2019-07-26:
@davids5 do you see why mavlink on USB doesn't connect here? The log looks correct apart from the missing SD card.

davids5, 2019-07-26:
@julianoes @airmind - nothing stands out. @airmind can you provide a copy of the dmesg output from the mac. With QGC running. Start with the FMU unplugged, then plug it in to USB. wait ~10 seconds then issue dmesg in a mac term window. The end of the log (last 11 seconds or so) is what I need to see.

airmind, 2019-07-27:
@davids5 

I logged 2 sessions of dmesg output. Are they what your need?

[dmesg_output1.txt](https://github.com/PX4/Firmware/files/3437974/dmesg_output1.txt)
[dmesg_output2.txt](https://github.com/PX4/Firmware/files/3437975/dmesg_output2.txt)

airmind, 2019-08-07:
@davids5 
Do you need a MindPX hardware to debug this issue? I can ship you one in a few days. 
This issue may or may not be platform specific, but a MindPX hardware can help to re-produce the problem for sure.

Just let me know.
Regards.

davids5, 2019-08-07:
@airmind - sorry for the delay. 

I am not sure just the flight controller will help. I have a older MAC mini....

I see nothing in the dmeg logs that I expected to see:

Here is FMU V2 plugged in
```
[1034016.754515] usb 2-1: new full-speed USB device number 8 using xhci_hcd
[1034016.922410] usb 2-1: New USB device found, idVendor=26ac, idProduct=0011
[1034016.926816] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[1034016.930968] usb 2-1: Product: PX4 BL FMU v2.x
[1034016.934130] usb 2-1: Manufacturer: 3D Robotics
[1034016.937567] usb 2-1: SerialNumber: 0
[1034017.112862] cdc_acm 2-1:1.0: ttyACM0: USB ACM device
[1034017.124244] usbcore: registered new interface driver cdc_acm
[1034017.127835] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[1034021.497186] usb 2-1: USB disconnect, device number 8
[1034022.911329] usb 2-1: new full-speed USB device number 9 using xhci_hcd
[1034023.078380] usb 2-1: New USB device found, idVendor=26ac, idProduct=0011
[1034023.082507] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[1034023.086628] usb 2-1: Product: PX4 FMU v2.x
[1034023.090572] usb 2-1: Manufacturer: 3D Robotics
[1034023.094395] usb 2-1: SerialNumber: 0
[1034023.100272] cdc_acm 2-1:1.0: ttyACM0: USB ACM device
```

Is this logged on the MAC in syslog?

airmind, 2019-08-09:
@davids5 
Do you mean system.log? No I did not see these message in /var/log/system.log
I got the chance to grab a basic Pixhawk for testing, and shows random results that sometimes it can connect, and sometimes can not. In either cases, ttyusbmodem01 has been created under /dev.
So seems to be a hardware platform independent issue.

What is the MACOS version you are running in Mac mini?

davids5, 2019-08-10:
@airmind it is old...
[image]

davids5, 2019-08-10:
@airmind 

>You can launch it from Spotlight, or by clicking on the Apple symbol in the menubar and selecting "About the Mac" and then clicking "System Report...".

>If you select USB from the list of hardware types on the left-hand side, you will get some information about the USB devices plugged into your Mac including VendorID, ProductID and Manufacturer.

What do you see under System report for USB devices?

airmind, 2019-08-12:
@davids5 

Here is what I saw using a basic Pixhawk.... it does not connect.

[image]

closed 2022-08-15 by bresch