PX4/PX4-Autopilot #11901: micro_rtps bridge fails to build

AlexisTM, 2019-04-24:
**Describe the bug**

If `fastrtpsgen` is in the PATH (and `FASTRTPSGEN_DIR` is not set), then the build fails with:

> generate_microRTPS_bridge.py: error: argument -f/--fastrtpsgen-dir: expected one argument

while the `generate_microRTPS_bridge.py` autodetects properly the presence of `fastrtpsgen`.

**To Reproduce**
1. Have `fastrtpsgen` in the PATH and `FASTRTPSGEN_DIR` not set.
2. Run `make px4_fmu-v3_rtps`

**Expected behavior**

The Firmware builds.

**Additional context**

This is due to the option `--fastrtpsgen-dir` to be set to `FASTRTPSGEN_DIR` in the CMakeLists.txt. Therefore if `FASTRTPSGEN_DIR` is not set, argparse fails. 

https://github.com/PX4/Firmware/blob/a9c3bce20da57e2fdd9c3ad630892506677ca377/src/modules/micrortps_bridge/micrortps_client/CMakeLists.txt#L76

TSC21, 2019-04-24:
This needs a condition here where it checks if `FASTRTPSGEN_DIR` is set and if not, looks on the default path. Will add that asap.

AlexisTM, 2019-04-25:
I have another solution, allowing no data for argparse, using `nargs='?'`

closed 2019-04-25 by TSC21