PX4/PX4-Autopilot #11903: How do we ensure smoothness of control signals?

RomanBapst, 2019-04-24:
**Issue**
Individual flight tasks may ensure smoothness but switching between them can cause steps in attitude & throttle setpoint.

**Ideas**
- Initialize flight task based on data from the old flight task such that no steps are produced -> need to consider quite a few corner cases
- Apply a blending for a certain amount of time -> easier to implement but not as clean as the first solution

**Expected results**
The attitude and throttle setpoint should be "continuous" all across a flight log which includes transitions between arbitrary modes.

dagar, 2019-04-24:
Can we go through a few cases an end user would experience and rank them?

 - `ManualPosition -> FlightTaskAuto` (flying in position mode and RTL engages)
 - transitions between different Auto FlightTasks?
 

### Structure to handle transitions
Switching to a new FlightTask is constructing an object in the same place the current FlightTasks resides. 

``` C++
int FlightTasks::_initTask(FlightTaskIndex task_index)
{
	switch (task_index) {
	case FlightTaskIndex::None:
		// already disabled task
		break;

	case FlightTaskIndex::ManualAltitude:
		_current_task.task = new (&_task_union.manualAltitude) FlightTaskManualAltitude();
		break;

	case FlightTaskIndex::ManualAltitudeSmooth:
		_current_task.task = new (&_task_union.manualAltitudeSmooth) FlightTaskManualAltitudeSmooth();
		break;

	case FlightTaskIndex::ManualPosition:
		_current_task.task = new (&_task_union.manualPosition) FlightTaskManualPosition();
		break;

       // ...............

	default:
		// invalid task
		return 1;
	}

	// task construction succeeded
	_current_task.index = task_index;
	return 0;
}
```

One idea would be to structure it such that we literally construct the new FlightTask from the old. From that point we could handle the general cases, but everything would also be in place to handle all the specific cases in as much detail as desired.

Example: switching to any Auto from Position control.

``` C++
FlightTaskAuto::FlightTaskAuto(const FlightTaskManualPosition& old_flight_task)
{
    // copy old setpoints, etc
}

```

hamishwillee, 2019-08-07:
I don't think this should be closed @bresch

closed 2022-08-15 by bresch