PX4/PX4-Autopilot #11905: RC loss failsafe doesn't Terminate

JMshare, 2019-04-24:
**Describe the bug**
The PWM outputs don't go to the failsafe values on RC lost with the Terminate failsafe action set up. 

**To Reproduce**
px4_fmu_v5
Using the current master (build 1.9.0 beta3)
Quadcopter DJI F450 airframe
Calibrate sensors, radio setup, safety setup set to Terminate on RC loss.
Reboot.
Arm in manual mode, turn off the radio, voila..

**Expected behavior**
According to the documentation, should output the Failsafe PWM values.

**Log Files and Screenshots**
[image]
[image]

**Additional context**
I get that turning off the quadcopter motors on RC loss may not be the best thing to do, but that is what I currently want to achieve. 
What else are the PWM_MAIN_FAIL parameters used for?
I get strange PWM values also using the standard plane configuration. Also I am not sure but it seems the behaviour kind of changes based on which flight mode am I in? Is there some magic happening that is not documented?

**Feature request**
The receiver usually has some RC loss failsafe setup, maybe there could be an option in PX4 to use these values is RC input?

julianoes, 2019-04-25:
My first though is that you need to actually enable the flight termination using this circuit breaker which is set (so the circuit is broken) by default:
https://dev.px4.io/en/advanced/parameter_reference.html#CBRK_FLIGHTTERM

JMshare, 2019-04-25:
Oh that works. 
However, for the period of RC loss timeout it unexpectedly accelerates the motors by around 50us which is a bit scary. The same thing happens if RC failsafe is set to disabled. Why is that happening?

Also sometimes if both RC loss and GPS loss are triggered at the same time, the flight doesn't recover after RC regained. I can arm and disarm but the motors won't spin. Have to reboot to regain control. Seems like the circuit breaker gets stuck in the triggered mode.

julianoes, 2019-04-25:
> However, for the period of RC loss timeout it unexpectedly accelerates the motors by around 50us which is a bit scary.

That's not optimal. Could you share a log please? Thanks.

> Have to reboot to regain control.

I think "termination" is supposed to be "terminal" :smile:.

JMshare, 2019-04-26:
> > However, for the period of RC loss timeout it unexpectedly accelerates the motors by around 50us which is a bit scary.
> 
> That's not optimal. Could you share a log please? Thanks.

I take that back, it had to do with me not flashing the firmware properly.

> > Have to reboot to regain control.
> 
> I think "termination" is supposed to be "terminal" 😄.

Not quite, if you have a RC loss failsafe setup to terminate it gives you all the control back when RC regained, actually it doesn't even disarm. Indeed, why would you want your vehicle to become unresponsive after you regained RC signal?
However, if you have RC loss failsafe set up to disabled, something kicks in and terminates the system for good saying 

> [commander] RC and GPS lost! Flight terminated.

 So after that you are not able to regain control even after RC is regained. 
[https://review.px4.io/plot_app?log=ef4da463-4199-46db-8fd0-c179704c4537](url)

In conclusion, why not to have an option to set the failsafe action called Failsafe, that would do what it does now as Terminate, and an action called Terminate, that would actually terminate? Besides of making things clearer, I wouldn't need to have the termination circuit breaker enabled.
And as a bonus there could also be an action called RC failsafe that would use the RC failsafe values. I thought that this might be the Disabled one, but it seems that the Disabled just holds the last attained values. Again, the documentation here doesn't help much.

julianoes, 2019-04-26:
Ok thanks for writing that up. I think I agree and this needs testing, fixing and documenting. It's on my todo list.

bresch, 2019-05-06:
Currently, the failsafe logic is not proper working. I revisited the termination part in PR #10179 
As shown in the diagram of that PR, termination for RC loss, data-link loss and geofence does not require to disable the circuit-breaker; it only needs to be configured via a parameter/QGC drop-down menus. 

Also, (the diagram does not show it yet but I will update it) termination from RC loss and data-link loss is recoverable while termination from FMU loss and geofence is no-return (requires desarm+reboot to recover).

The `[commander] RC and GPS lost! Flight terminated.` is an old piece of code that only checks for RC and does not care about the failsafe parameter (in  current master, if you disable the circuit-breaker, a RC loss or a data-link loss triggers termination...)

JMshare, 2019-05-06:
> Currently, the failsafe logic is not proper working. I revisited the termination part in PR #10179
> As shown in the diagram of that PR, termination for RC loss, data-link loss and geofence does not [...]

Can anyone test this? It didn't work for me without disabling the circuit breaker.
Seems like a serious issue to me, given that it's not even legal to fly without the RC failsafes set up properly (at least in UK the motors must be shut).

JMshare, 2019-05-06:
I tested it again and it seems ok in the current master. I must have been using some older build before.

bresch, 2019-05-06:
> Can anyone test this?

It works in my PR, I tested. I also tested on master and I have the same results as you have: it doesn't work.

bresch, 2019-05-08:
@JMshare I just tested on master and it does **not** work on current master without disabling the circuit breaker. However #11982 fixes `[commander] RC and GPS lost! Flight terminated.` trigger when not configured.
As a quick workaround (before we merge #10179) , you can use the "lockdown" failsafe action for RC loss. This doesn't apply the failsafe values, but the disarmed ones. This doesn't require to change any circuit-breaker.

bresch, 2020-03-05:
Fixed by https://github.com/PX4/Firmware/pull/10179

closed 2020-03-05 by bresch