PX4/PX4-Autopilot #11912: High sensor noise in SITL when disarmed

julianoes, 2019-04-25:
I realized that in SITL Gazebo (using `standard_vtol`) the sensor noise (especially accel) is quite high.

In my opinion this is far from reality and should be improved.

I stumbled upon this in SDK tests where arming failed with the message:
```
[commander] Preflight Fail: Accel Range, hold still on arming
```
https://logs.px4.io/plot_app?log=01919d08-3298-45ff-8cb2-c99929b890ca

@bresch also saw it in SITL CI.

LorenzMeier, 2019-04-25:
I'm making this a release broker since it will induce SITL flakyness that should not be in a release.

julianoes, 2019-04-25:
So while we don't scale noise with motor speeds or the fact that we're armed, this actual issue seems to be caused by the `standard_vtol` bouncing on the ground or somehow interacting with it.

julianoes, 2019-04-25:
I'll push a fix tomorrow, I've managed to get the `standard_vtol` less jittery and added somewhat proper airspeed noise.

PR #11917 (2019-04-28): Fix standard_vtol vibration on ground
This addresses the issue where the `standard_vtol` (delta quad) vibrates on the ground.

Also, this moves the airspeed noise from `simulator_mavlink.cpp` to the Gazebo side.

Fixes #11912, fixes #11213.
closed 2019-04-28 by dagar