PX4/PX4-Autopilot #11916: When disabling GPS_RAW_INT or ATTITUDE_QUATERNION via MAV_CMD_SET_MESSAGE_INTERVAL all streams get disabled instead of just one

jdzerve, 2019-04-26:
**Description**
I'm using MAV_CMD_SET_MESSAGE_INTERVAL to disable various telemetry streams from px4. 
Everything works as expected except when GPS_RAW_INT is disabled. PX4 stops sending any telemetry at all. This is reproducable on both SITL and on Pixhawk.

On pixhawk all telemetry can be disabled by disabling GPS_RAW_INT via setting rate to -1 in MAV_CMD_SET_MESSAGE_INTERVAL command.

On SITL the same is achieved when sending that command to gcs port $udp_gcs_port_local.
On $udp_offboard_port_local the same is achieved by disabling ATTITUDE_QUATERNION stream. 

**Expected behavior**
After disabling ATTITUDE_QUATERNION stream PX4 should stop sending only ATTITUDE_QUATERNION, not all messages.

**Drone (please complete the following information):**
- SITL generic quad.

**Additional info**
Fortunately this applies to latest master and beta branches only. "stable" branch works as expected.

julianoes, 2019-07-26:
Thanks for raising that issue. This indeed sounds like a bug.

julianoes, 2019-07-26:
@jdzerve I tried this in SITL and I actually can't reproduce it. I stopped `ATTITUDE_QUATERNION` and all the other messages are still arriving.

julianoes, 2019-07-26:
@jdzerve could you check master again? I tested 71613ac631871efdf7d2818f085ecff5ab5037ea.

MikeCharikov, 2019-10-04:
Hi! It is reproduced on current stable (1.9.2), but it looks like it is fixed in master.

closed 2019-10-07 by julianoes