PX4/PX4-Autopilot #11919: PX4 SITL/mavros CMD: Unexpected command 22, result 0

mahdinobar, 2019-04-26:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

I use MAVROS node to translate MAVLink messages into ROS messages so I  can communicate between ROS and Gazebo through PX4. The problem is when the commander offboard control phase finishes, I receive the warning **CMD: Unexpected command 22, result 0** then the drone lands, whereas I should continue the flight based on the velocity command from my ROS node. So my question is what does this warning message mean and how it can be corrected?

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: ?Kinetic?
Ubuntu: ?16.04?

### Autopilot type and version

PX4

Version: ?v1.8.2?

julianoes, 2019-04-29:
Are you continuously sending the offboard commands or do you stop sending them?
The unexpected command 22 is: https://mavlink.io/en/messages/common.html#MAV_CMD_NAV_TAKEOFF

julianoes, 2019-07-30:
@mahdinobar could you follow up here please?

julianoes, 2019-10-28:
I'm going to close this because there was no follow up.

closed 2019-10-28 by julianoes