PX4/PX4-Autopilot #11924: require external mag (optionally)

dagar, 2019-04-28:
I've noticed a number of setups that manage to slip through without an external mag, despite the user thinking one is present.

Ideas
 1. Add a configurable preflight check that requires an external mag by default, but can be easily disabled.
 2. Start explicitly configuring all external devices rather than probing for everything. By requiring the full configuration to be entered we have all the necessary information.

dagar, 2019-04-28:
* Required sensors
    * board manifest
    * user configured externals
    * preflight checks, useful user facing errors if missing
 * Optional sensors
    * does this category even need to exist?

ryanjAA, 2019-05-08:
This would be really good.