PX4/PX4-Autopilot #11929: "uorb status" command hangs firmware on pixhawk 4

a6a3uh, 2019-04-29:
**Describe the bug**
On pixhawk 4 running `uorb status` in MAVLink console hangs firmware.

**To Reproduce**
Tested with latest master c50c44cc582956aefd32e39a6c7a5210ba4e1585
Steps to reproduce the behaviour:
1. Drone switched on '
2. Connect with QGC via telemetry link or directly via a USB cable.
3. Got to MAVLink console
4. run `uorb status`
5. see link disconnected, only reboot helps

**Expected behavior**
I expect to see a full list of topics when running `uorb status`. But I see only a couple of them before firmware hangs.

julianoes, 2019-04-30:
@a6a3uh thanks for reporting. I can reproduce this.

bkueng, 2019-04-30:
Fix is here: https://github.com/PX4/Firmware/pull/11941

closed 2019-05-02 by julianoes