PX4/PX4-Autopilot #11932: vl53lxx driver is currently only for vl53l0x

bys1123, 2019-04-29:
**Describe the bug**
vl53lxx driver is currently only for vl53l0x, also vl53l1x shouldn't share the same driver. They have 
very different register addresses.

**Reference**
Arduino driver:
https://github.com/pololu/vl53l0x-arduino/blob/master/VL53L0X.h
https://github.com/pololu/vl53l1x-arduino/blob/master/VL53L1X.h
ArduPilot driver:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_RangeFinder/AP_RangeFinder_VL53L0X.cpp
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_RangeFinder/AP_RangeFinder_VL53L1X.h

**Related open issue/PR**
#11842 #11896

mcsauder, 2019-04-30:
@bys1123 , I've pushed up a commit in PR #11896 that has a shot at functioning for your needs.  I have now hardware to test with, so you might need to help with debugging.

I've added register maps for the vl530x and vl531x sensors in respective *.h files.  Please find me on slack if you can help me with testing this work.

bys1123, 2019-04-30:
@mcsauder Thanks! I'll test it today!

bys1123, 2019-04-30:
@DanielePettenuzzo We need your help.

mcsauder, 2019-04-30:
The hard coded register values in the vl53lxx driver present a problem in trying to get dual functionality for the vl530x and vl531x drivers out of a single implementation. STMicro doesn't publish a register map for the vl530x sensor, and the API doesn't provide names for many of the hard-coded values, so it isn't easy to identify what exactly is taking place.  It looks like we'll need a new implementation for the vl531x, so I'll do some more work to see how easily this can be accomplished.  In the meantime I've reverted PR #11896 and will work on this feature in a new PR!

mcsauder, 2019-05-08:
@bys1123 , I have a WIP branch here in case you can use the register maps I have condensed out of ST's APIs: https://github.com/PX4/Firmware/compare/master...mcsauder:vl53l1x_driver_creation

mcsauder, 2019-08-07:
@dagar created PR #12575 for this functionality, but I have not had a chance to test it yet.

zwn, 2019-10-19:
What is the current status of VL53L1x support? If I want to help, where should I go?

TSC21, 2019-10-19:
@zwn any progress should be followed in https://github.com/PX4/Firmware/pull/12575.

bys1123, 2020-04-19:
#12575 is closed, waiting new PR.