PX4/PX4-Autopilot #11936: Mavlink version control per instance

ryanjAA, 2019-04-29:
It would be helpful to have more granular control over which version of mavlink is going over which stream, i.e. telem 1 as Mavlink v2, telem 2 as Mavlink v1, etc. This is simply to give specific control over things you may want to integrate in. Something parameter based makes the most sense. I imagine you could achieve this from command line or startup script but this would be easier overall.

@dagar

dagar, 2019-04-29:
I'll update the existing `MAV_PROTO_VER` to be per parameter like `MAV_x_MODE`, `MAV_x_RATE`, etc.

`MAV_PROTO_VER -> MAV_x_PROTO_VER`

ryanjAA, 2019-04-29:
That would be great! Thanks

sampathkumarch, 2019-05-01:
i have flight data from from pixhawk in the for
mate have ulg. that  formate how can convert excel please explain detail

alexcherpi, 2019-05-21:
Hi, it would be great to stop specific instance. Now there is only the 'mavlink stop-all' command which stop all instance.

ryanjAA, 2020-09-11:
Did this ever get any movement? Or is there a workaround?