PX4/PX4-Autopilot #11942: Unable to biuld px4 v 1.9.0 on beagle bone blue native

7st4r, 2019-04-30:
**Describe the bug**
A clear and concise description of the bug.

**To Reproduce**
Steps to reproduce the behavior:
1. Drone switched on '...'
2. Uploaded mission '....' (attach QGC mission file)
3. Took off '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/en/getting_started/flight_reporting.html)).
- Share the link to a log showing the problem on [PX4 Flight Review](http://logs.px4.io/).

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
- Describe the type of drone.
- Photo of the IMU / autopilot setup if possible.

**Additional context**
Add any other context about the problem here.

7st4r, 2019-04-30:
Here is a log of error massage;

debian@beaglebone:/var/lib/cloud9/test/Firmware$ make beaglebone_blue_native
-- PX4 config file: /var/lib/cloud9/test/Firmware/boards/beaglebone/blue/native.cmake
-- PX4 config: beaglebone_blue_native
-- PX4 platform: posix
-- PX4 lockstep: disabled
-- PX4 version: v1.9.0-rc0
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 6.3.0
-- The C compiler identification is GNU 6.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PythonInterp: /usr/bin/python (found version "2.7.13")
-- Found PY_jinja2: /usr/lib/python2.7/dist-packages/jinja2
DF Drivers: mpu9250;bmp280
Adding DF driver: mpu9250
Adding DF driver: bmp280
-- PX4 ECL: Very lightweight Estimation & Control Library v0.9.0-676-gc4492b1
-- Configuring done
-- Generating done
-- Build files have been written to: /var/lib/cloud9/test/Firmware/build/beaglebone_blue_native
Scanning dependencies of target df_bmp280
Scanning dependencies of target df_mpu9250
Scanning dependencies of target df_driver_framework
Scanning dependencies of target uorb_headers
[  0%] Building CXX object src/lib/DriverFramework/drivers/bmp280/CMakeFiles/df_bmp280.dir/BMP280.cpp.o
[  0%] Building CXX object src/lib/DriverFramework/drivers/mpu9250/CMakeFiles/df_mpu9250.dir/MPU9250.cpp.o
[  0%] Generating uORB topic headers
[  0%] Building CXX object src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/DriverFramework.cpp.o
[  0%] Building CXX object src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/WorkMgr.cpp.o
[  0%] Building CXX object src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/WorkItems.cpp.o
/var/lib/cloud9/test/Firmware/src/lib/DriverFramework/drivers/bmp280/BMP280.cpp: In member function ‘int DriverFramework::BMP280::loadCalibration()’:
/var/lib/cloud9/test/Firmware/src/lib/DriverFramework/drivers/bmp280/BMP280.cpp:124:58: error: cast from ‘uint8_t* {aka unsigned char*}’ to ‘uint16_t* {aka short unsigned int*}’ increases required alignment of target type [-Werror=cast-align]
  uint16_t *calib_value_ptr = (uint16_t *) &calib_values[0];
                                                          ^
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors
src/lib/DriverFramework/drivers/bmp280/CMakeFiles/df_bmp280.dir/build.make:62: recipe for target 'src/lib/DriverFramework/drivers/bmp280/CMakeFiles/df_bmp280.dir/BMP280.cpp.o' failed
make[3]: *** [src/lib/DriverFramework/drivers/bmp280/CMakeFiles/df_bmp280.dir/BMP280.cpp.o] Error 1
CMakeFiles/Makefile2:1484: recipe for target 'src/lib/DriverFramework/drivers/bmp280/CMakeFiles/df_bmp280.dir/all' failed
make[2]: *** [src/lib/DriverFramework/drivers/bmp280/CMakeFiles/df_bmp280.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....

7st4r, 2019-04-30:
debian@beaglebone:/var/lib/cloud9/test/Firmware$ make beaglebone_blue_native
-- PX4 config file: /var/lib/cloud9/test/Firmware/boards/beaglebone/blue/native.cmake
-- PX4 config: beaglebone_blue_native
-- PX4 platform: posix
-- PX4 lockstep: disabled
-- PX4 version: v1.9.0-rc0
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 6.3.0
-- The C compiler identification is GNU 6.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PythonInterp: /usr/bin/python (found version "2.7.13")
-- Found PY_jinja2: /usr/lib/python2.7/dist-packages/jinja2
DF Drivers: mpu9250;bmp280
Adding DF driver: mpu9250
Adding DF driver: bmp280
-- PX4 ECL: Very lightweight Estimation & Control Library v0.9.0-676-gc4492b1
-- Configuring done
-- Generating done
-- Build files have been written to: /var/lib/cloud9/test/Firmware/build/beaglebone_blue_native
Scanning dependencies of target df_bmp280
Scanning dependencies of target df_mpu9250
Scanning dependencies of target df_driver_framework
Scanning dependencies of target uorb_headers
[  0%] Building CXX object src/lib/DriverFramework/drivers/bmp280/CMakeFiles/df_bmp280.dir/BMP280.cpp.o
[  0%] Building CXX object src/lib/DriverFramework/drivers/mpu9250/CMakeFiles/df_mpu9250.dir/MPU9250.cpp.o
[  0%] Generating uORB topic headers
[  0%] Building CXX object src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/DriverFramework.cpp.o
[  0%] Building CXX object src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/WorkMgr.cpp.o
[  0%] Building CXX object src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/WorkItems.cpp.o
/var/lib/cloud9/test/Firmware/src/lib/DriverFramework/drivers/bmp280/BMP280.cpp: In member function ‘int DriverFramework::BMP280::loadCalibration()’:
/var/lib/cloud9/test/Firmware/src/lib/DriverFramework/drivers/bmp280/BMP280.cpp:124:58: error: cast from ‘uint8_t* {aka unsigned char*}’ to ‘uint16_t* {aka short unsigned int*}’ increases required alignment of target type [-Werror=cast-align]
  uint16_t *calib_value_ptr = (uint16_t *) &calib_values[0];
                                                          ^
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors
src/lib/DriverFramework/drivers/bmp280/CMakeFiles/df_bmp280.dir/build.make:62: recipe for target 'src/lib/DriverFramework/drivers/bmp280/CMakeFiles/df_bmp280.dir/BMP280.cpp.o' failed
make[3]: *** [src/lib/DriverFramework/drivers/bmp280/CMakeFiles/df_bmp280.dir/BMP280.cpp.o] Error 1
CMakeFiles/Makefile2:1484: recipe for target 'src/lib/DriverFramework/drivers/bmp280/CMakeFiles/df_bmp280.dir/all' failed
make[2]: *** [src/lib/DriverFramework/drivers/bmp280/CMakeFiles/df_bmp280.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
[  0%] Building CXX object src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/DevMgr.cpp.o
[  0%] Building CXX object src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/DevObj.cpp.o
[  0%] Building CXX object src/lib/DriverFramework/drivers/mpu9250/CMakeFiles/df_mpu9250.dir/MPU9250_mag.cpp.o
[  0%] Building CXX object src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/SyncObj.cpp.o
[  1%] Building CXX object src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/DFList.cpp.o
[  1%] Building CXX object src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/Time.cpp.o
[  1%] Linking CXX static library libdf_mpu9250.a
[  1%] Built target df_mpu9250
[  1%] Building CXX object src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/DFDiag.cpp.o
[  1%] Building CXX object src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/I2CDevObj.cpp.o
[  1%] Building CXX object src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/SPIDevObj.cpp.o
[  1%] Building CXX object src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/ImuSensor.cpp.o
[  1%] Building CXX object src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/BaroSensor.cpp.o
[  1%] Building CXX object src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/MagSensor.cpp.o
[  2%] Linking CXX static library libdf_driver_framework.a
[  2%] Built target df_driver_framework
[  2%] Built target uorb_headers
Makefile:160: recipe for target 'all' failed
make[1]: *** [all] Error 2
Makefile:190: recipe for target 'beaglebone_blue_native' failed
make: *** [beaglebone_blue_native] Error 2

dagar, 2019-04-30:
It's the cast align warning (`-Werror=cast-align`). What we really need to do is update this build and get it in the CI system. I'd like to either bring librobotcontrol in as a submodule or via cmake ExternalProject. We should also be able to dump the native/cross configs. A single default can work for both with the architecture specified.

julianoes, 2019-05-02:
You can work around a warning by removing `-Werror` for now here:
https://github.com/PX4/Firmware/blob/b35d0485669fd47173eb742bc0a08cb0dbac81f0/cmake/px4_add_common_flags.cmake#L66

closed 2019-08-15 by stale[bot]