PX4/PX4-Autopilot #11945: px4_atomic breaks snapdragon flight

smnogar, 2019-05-01:
The inclusion of [px4_atomic](https://github.com/PX4/Firmware/commit/b6e2ac74deb1b1d2600593d0fdde2cee361f4615)  breaks the qurt side of the snapdragon flight.

It still builds on both the posix and qurt side, but uorb fails on the DSP side with the following errors in nano-dm:

```
1573827: HAP:20543:undefined PLT symbol __atomic_store_1 (1210) /libpx4.so (symbol.c:303)
1573827: HAP:20543:undefined PLT symbol __atomic_store_4 (1211) /libpx4.so (symbol.c:303)
1573827: HAP:20543:undefined PLT symbol __atomic_load_4 (445) /libpx4.so (symbol.c:303)
1573827: HAP:20543:undefined PLT symbol __atomic_load_1 (444) /libpx4.so (symbol.c:303)
1573828: HAP:20543:undefined PLT symbol _LSin (705) /libpx4.so (symbol.c:303)
```

Looks like some sort of linker error. Previous commits in version 1.9 or in 1.8.2 work successfully, so its not an issue of the DSP firmware being out of date.

My px4.config file is only:

```
uorb start
```

julianoes, 2019-05-02:
@smnogar thanks for raising this.

julianoes, 2019-05-06:
Ok, I can reproduce this issue:
```
   1461347: HAP:63:undefined PLT symbol __atomic_store_1 (1210) /libpx4.so (symbol.c:303)
   1461347: HAP:63:undefined PLT symbol __atomic_store_4 (1211) /libpx4.so (symbol.c:303)
   1461347: HAP:63:undefined PLT symbol __atomic_load_4 (445) /libpx4.so (symbol.c:303)
   1461347: HAP:63:undefined PLT symbol __atomic_load_1 (444) /libpx4.so (symbol.c:303)
   1461348: HAP:63:undefined PLT symbol _LSin (705) /libpx4.so (symbol.c:303)
```

julianoes, 2019-05-06:
I was able to fix `__atomic_store` and `__atomic_load` by ifdefing the usage, I will create a PR soon. However, I haven't resolved `_LSin` yet.

closed 2019-05-06 by dagar

smnogar, 2019-05-06:
Thanks everyone!

PR #11968 (2019-05-06): Fix Snapdragon linking
This fixes two linking issues on Snapdragon. For more infos check out the commit messages.

The fix for the missing atomic builtin was fairly obvious, for the `_LSin` symbol I had to bisect and then remove `sih` from the Snappy build.

Fixes #11945.