PX4/PX4-Autopilot #11952: Stabilized but RC components is yaw dependent

alexcherpi, 2019-05-03:
In manual mode, when I monitor the output (_actuators.control[]) of the attitude stabilization of my quadcopter (run in mc_att_control_main.cpp) I see that for a constant roll RC input command the _actuators.control[0] and _actuator_control[1] changes when I the quadcopter in yaw (by hand).  For example when I turn by 180° in yaw the _actuators_control[0] get a minus sign... Why?

This seems to be as if the RC commands were given in the World Reference Frame. But in manual mode this shouldn't be the case. 

I am using PX4 v1.8.0 with the pixracer board but I am disabling the magnetometers.
Thanks

MaEtUgR, 2019-05-07:
**Background**
@alexcherpi Correct me if I misunderstood you. How PX4 currently does it is it generates a complete 3D attitude setpoint orientation including yaw from your sticks https://github.com/PX4/Firmware/blob/72b898fbf6d9f61c60da872dabed66a82974aaea/src/modules/mc_att_control/mc_att_control_main.cpp#L413-L443. The yaw is generated by moving the absolute yaw around with the rate given by the stick here: https://github.com/PX4/Firmware/blob/72b898fbf6d9f61c60da872dabed66a82974aaea/src/modules/mc_att_control/mc_att_control_main.cpp#L421
After that the [attitude controller](https://github.com/PX4/Firmware/tree/72b898fbf6d9f61c60da872dabed66a82974aaea/src/modules/mc_att_control/AttitudeControl) gets called to execute on that setpoint: https://github.com/PX4/Firmware/blob/72b898fbf6d9f61c60da872dabed66a82974aaea/src/modules/mc_att_control/mc_att_control_main.cpp#L521 The controller will always take the "shortest path" to reach the desired attitude but will prioritize roll and pitch over yaw with the yaw weight applied here https://github.com/PX4/Firmware/blob/72b898fbf6d9f61c60da872dabed66a82974aaea/src/modules/mc_att_control/AttitudeControl/AttitudeControl.cpp#L84.

Note that the strategy of the algorithm didn't change for a long time (not new in 1.8) the only thing that changed in 2017 is the quaternion attitude controller executing the setpoint https://github.com/PX4/Firmware/pull/8003.

**Answer**
I would say that should explain why when you turn your vehicle in a bench test it will try to reach back to the original absolute yaw it locked to and not just correct roll. This algorithm strategy things **has the big advantage that the yaw angle will not drift** over time given your yaw estimate is correct and doesn't drift (which it might not be anymore without magnetometer). As a result you can fly very straight lines in manual and usually don't have to touch the yaw stick. I assume what you expected to see is only roll and pitch being controlled in an absolute manner and for yaw only the rate controller being ran.

> This seems to be as if the RC commands were given in the World Reference Frame. But in manual mode shouldn't it be in Body Reference Frame?

**Conclusion**
I'm open to finding an appropriate solution for example an option for your use case or eventually after in-depth testing convince myself of a better default way to handle yaw in manual. Could you explain what your goal is? That would help to understand the requirements.

**Ideas**
@bresch Just told me about the use case where you have a vehicle that has very big yaw errors while flying normally or even doesn't care about the yaw at all and has disabled yaw control. In that case correcting roll pitch according to a setpoint which assumes a different yaw angle to be locked than the vehicle actually has is totally confusing. Is that your use case?

There are multiple actions we can take:
- (Optionally) switch stick setpoint generation for roll and pitch to the current yaw vehicle state and test the real world accuracy difference
- Expose the yaw weight of the attitude controller such that reducing significance of or even disabling yaw tracking is easily possible without screwing up the output gains (that is anyways on my todo list).
- (Optionally) switch yaw control to rate control only for testing. That would get rid of any yaw lock and just try to reduce drift.

alexcherpi, 2019-05-07:
Dear MaEtUgR,

Problem solved! But thank you for your complete answer!
Actually I rebased my branch from v1.8.0 to v1.8.2 and the issue disappeared.
The cause are possibly the following: issue in v1.8.0 OR mess in my code (I would go for the 2nd option).

alexcherpi, 2019-05-07:
Problem solved! But thank you for your complete answer!
Actually I rebased my branch from v1.8.0 to v1.8.2 and I changed my custom mixer to the 'Generic Quadrotor X' mixer and the issue disappeared.
The cause are possibly the following: issue in v1.8.0 / mess in my code / mess with my custom mixer.
Sorry if it cannot help PX4 to be improved but thanks for your help

MaEtUgR, 2019-05-09:
Nice to hear! Thanks for the update and feel free to reopen and ask back if you have further questions or ideas. I will close this issue now.

bresch, 2019-05-13:
@alexcherpi I don't understand why you don't see the issue anymore, the logic did not change and as @MaEtUgR  explained, there is a reason why it is like that right now.
There is a workaround for VTOL planes that have bad yaw tracking here:
https://github.com/PX4/Firmware/blob/90758258723f837dcadf1d4c303229a722f64ec9/src/modules/mc_att_control/mc_att_control_main.cpp#L455-L490

Maybe we can put a parameter instead of `if(is_vtol)` so you can also use it with any "free to yaw" vehicle.

alexcherpi, 2019-05-17:
Hi bresch, the cause are possibly the following: 
- issue in v1.8.0 (doesn't seems to be the case)
- mess with my custom mixer (normally not as it is used after creating the _actuators.control[] uorb message)
- mess in my code
So I think the poblem was a mess in my code, sorry if it cannot help PX4 to be improved.

closed 2019-07-09 by bresch