PX4/PX4-Autopilot #11955: Add model of Aion R1 rover to Gazebo

ItsTimmy, 2019-05-03:
**Describe problem solved by the proposed feature**
The [Aion R1 Rover](http://docs.aionrobotics.com/en/latest/r1-ugv.html) is being used as a development platform for rover support in PX4. The current rover model in Gazebo does not accurately reflect this rover.

**Describe your preferred solution**
Add an accurate model of the Aion R1 rover for testing in Gazebo.

**Additional context**
The existing rover model in Gazebo is a Polaris golf cart.

julianoes, 2019-06-13:
Found this: https://github.com/aionrobotics/r1_ugv_sim

TSC21, 2019-09-11:
@ItsTimmy is this still on your TODO?

Jaeyoung-Lim, 2019-09-26:
I will take a look at this this week end. I don't think it will take too long to do. However, @julianoes the link you have doesn't have a license. Would it be okay to adapt the xacro files into sdf and have it it `sitl_gazebo` ?

julianoes, 2019-09-26:
> the link you have doesn't have a license

I don't know but I would copy it and add a reference to where it's from in a comment. I would hope aionrobotics are ok that we use and thus advertise their product in our sim.

weiLiangHare, 2020-01-11:
I have the same trouble. Can anyone solve it

Jaeyoung-Lim, 2020-01-11:
@weiLiangHare Which issue are you having? This is not a bug, but a place holder for possible enhancement

weiLiangHare, 2020-01-11:
I hope sitl_gazebo can support Aion R1 rover simulation.I tried to add it myself.But there are problems。I work on the this branch [https://github.com/PX4/Firmware/tree/pr-sitl-r1](url),and my sdf is here。It can work, but it is not ideal.  @Jaeyoung-Lim 
```
<?xml version="1.0" ?>
<sdf version="1.5">
  <model name="aion_robotics_r1_rover">
    <link name="base_link">
      <collision name="collision">
        <!-- <pose>0 0 -0.3 0 0 0</pose> -->
        <pose>0 0 0.1 0 0 0</pose>
        <geometry>
          <box>
            <size>1.0074 0.5709 0.2675</size>
          </box>
        </geometry>
      </collision>
      <inertial>
        <mass>33.855</mass>
        <pose>-0.0856132 -0.000839955 0.238145 0 0 0</pose>
        <inertia>
          <ixx>2.2343</ixx>
          <ixy>-0.023642</ixy>
          <ixz>0.275174</ixz>
          <iyy>3.42518</iyy>
          <iyz>0.00239624</iyz>
          <izz>2.1241</izz>
        </inertia>
      </inertial>
 
      <visual name="base">
        <pose>0 0 0 0 0 0</pose>
        <geometry>
          <mesh>
            <uri>model://aion_robotics_r1_rover/meshes/base_link.stl</uri>
          </mesh>
        </geometry>
        <material>
          <script>
            <name>Gazebo/FlatBlack</name>
          </script>
        </material>
      </visual>
      <!-- Top Plate -->
      <visual name="top_plate">
        <pose>0 0 0 0 0 0</pose>
        <geometry>
          <mesh>
            <uri>model://aion_robotics_r1_rover/meshes/top_plate.stl</uri>
          </mesh>
        </geometry>
        <material>
          <script>
            <name>Gazebo/Yellow</name>
          </script>
        </material>
      </visual>
      <!-- Front Bumper -->
      <visual name="front_bumper">
        <pose>0.47 0 0.091 0 0 0</pose>
        <geometry>
          <mesh>
            <uri>model://aion_robotics_r1_rover/meshes/bumper.stl</uri>
          </mesh>
        </geometry>
        <material>
          <script>
            <name>Gazebo/FlatBlack</name>
          </script>
        </material>
      </visual>
      <!-- Rear Bumper -->
[... 422 lines cut ...]
        </channel>
       <channel name="back_right_wheel_drive">
          <input_index>1</input_index>
          <input_offset>0</input_offset>
          <input_scaling>20</input_scaling>
          <zero_position_disarmed>0</zero_position_disarmed>
          <zero_position_armed>0</zero_position_armed>
          <joint_control_type>velocity</joint_control_type>
          <joint_name>back_right_joint</joint_name>
          <joint_control_pid>
            <p>10</p>
            <i>1</i>
            <d>0</d>
            <iMax>800</iMax>
            <iMin>-800</iMin>
            <cmdMax>1200</cmdMax>
            <cmdMin>-1200</cmdMin>
          </joint_control_pid>
        </channel>

     </control_channels>
    </plugin>
  </model>
</sdf>
```

Jaeyoung-Lim, 2020-01-11:
@weiLiangHare Cool video! I have added my WIP progress of the rover in https://github.com/PX4/sitl_gazebo/pull/394 , but I was having some problems on making the wheel dynamics correct.

weiLiangHare, 2020-01-12:
@Jaeyoung-Lim  Cool,good job.Am looking forward to

FaboNo, 2020-04-02:
@Jaeyoung-Lim Is it possible to know the progress of the R1 integration in gazebo? Indeed because of the "situation", all of us are remotely working and thus I will investigate the simulation in order to move forward.  So I will test the L1 controller in simulation.
One more question, I guess that if PX4 is correctly configured the "wiring" with R1 motors is done automatically?
Thank you

Jaeyoung-Lim, 2020-04-02:
@FaboNo Currently there is not much progress, please feel free to move it forward. Currently the wiriing is already done with this PR. It's just with the current configuration of contact parameters the vehicle does not turn

FaboNo, 2020-04-02:
@Jaeyoung-Lim oh I see but when I look at the video from @weiLiangHare looks like it can turn no? I will take the sdf file from @weiLiangHare  and do some tests.

Jaeyoung-Lim, 2020-04-02:
@FaboNo That would be a lot of help, thanks!

Jaeyoung-Lim, 2020-04-13:
@FaboNo Finally got this running:

[image]

FaboNo, 2020-04-13:
@Jaeyoung-Lim Awesome!!! 
On my side I have to make px4 and sitl gazebo working and then connect with ROS - I will follow the "px4/obstacle" instructions .
Just few questions: 
When the PR will be approved the r1 rover will be added in stil_gazebo/models? This is a question I will ask about sitl_gazebo, is how to add a new model...
Looks like the L1 controller is fully working now (besides moving backward), so it is able to handle lon/lat coordinates as well as cartesian coordinates (in case there is a SLAM in the loop) - I guess the answer is yes

Jaeyoung-Lim, 2020-04-13:
@FaboNo I have linked the PR of the sitl_gazebo in #14652

closed 2020-05-05 by Jaeyoung-Lim

ThomasRegis, 2024-07-03:
When I use the aion r1_rover sdf model, I have troubles : 
- The car is moving alone ( maybe inertial issue ? )
- I try to add Ardupilot plugins but it doesn't work ! 
Is there someone who could help me please ?

Jaeyoung-Lim, 2024-07-03:
@ThomasRegis Please don't hijack the topic of this issue. If you have a new issue, please consider posting the description of your issue and instructions on how to reproduce it.