PX4/PX4-Autopilot #11956: Update `gnd_pos_control` and `ground_att_control` modules

ItsTimmy, 2019-05-03:
**Describe problem solved by the proposed feature**
The current rover position and attitude control modules are based on fixed-wing control. They work for simple waypoint navigation, but should be updated to account for different types of steering (Ackerman vs. Differential steering) and limitations of rovers. Additionally, rovers behave strangely around edge cases, such as when the `navigator` module would instruct a fixed-wing craft to loiter, and when instructing the rover to "take off" or "land".

**Describe your preferred solution**
Research existing control algorithms for Ackerman and Differential steering vehicles, and implement them in the `gnd_pos_control` and `gnd_att_control` modules.

**Describe possible alternatives**
The existing modules implement Ackerman control, so a new separate module could be created for differential steering.

julianoes, 2019-05-16:
From call:
@ItsTimmy: differential controller spins in circle when waypoint reached.
@dagar: differential steering rover controller could work like multicopter controller. We could have two controllers, one for Ackermann, one for differential steering.

SalimTerryLi, 2019-05-24:
Can we add another parameter to determine whether the rover should do in-situ turning or directly move forward while do steering?
I' m looking forward to a solution which can dynamically switch between Ackermann & differential steering. Because I want my rover can have a fluent behavior in wide area but also can easily adapt into limited space. (with the help of RTK)
We can build a simple framework in gnd_position_control at first and then finish the codes for differential steering?
Or, any better ways?

julianoes, 2019-06-06:
@ItsTimmy can you make a pull request for this change?

ItsTimmy, 2019-06-11:
I have created a WIP pull request: #12239 . However, this depends upon #11918 , so it must be merged first.

PR #12239 (2019-07-11): Rewrote gnd_pos_control and removed gnd_att_control
**Describe problem solved by the proposed pull request**
The original implementation of ground rover control was based on fixed wing control, and so included unnecessary code for attitude control. This caused some problems with edge cases, such as loitering and return to launch for rovers.

Resolves #11956

**Test data / coverage**
This was tested both in SITL with `gazebo_rover` and on the Aion R1 rover.

**Describe your preferred solution**
The L1 controller outputs a desired turn radius. With a fixed-wing craft, this is used to calculate a [... cut]
closed 2019-07-11 by dagar