PX4/PX4-Autopilot #11957: Add support for rover wheel encoders in PX4

ItsTimmy, 2019-05-03:
**Describe problem solved by the proposed feature**
Many rovers, including the (Aion R1)[http://docs.aionrobotics.com/en/latest/r1-ugv.html] which is being used for development, incorporate wheel encoders. These could play a similar role to optical flow and RTK sensors, in providing local, small-scale velocity feedback.

**Describe your preferred solution**
Create a driver which reads wheel encoder data from the (BasicMicro RoboClaw)[http://downloads.ionmc.com/docs/roboclaw_user_manual.pdf] motor driver on the Aion R1, and publishes it as a uORB topic. Create a module which uses this data for navigation.

**Describe possible alternatives**
The existing optical flow module could be used for similar navigation purposes, but calibrating the sensors properly might be difficult due to how low they would be to the ground.

oneWayOut, 2019-05-09:
According to https://discuss.px4.io/t/pixhawk-enabling-gpio-output/2457 , It's possible to read wheel encoders signal directly from GPIO pins.
Ardupilot have already implemented this, http://ardupilot.org/rover/docs/wheel-encoder.html#wheel-encoder

julianoes, 2019-08-05:
The roboclaw driver https://github.com/PX4/Firmware/pull/12416 has been merged but more work would be required to fuse this into the estimator.

julianoes, 2019-08-05:
The estimator part is tracked in #12350, closing.

closed 2019-08-05 by julianoes