PX4/PX4-Autopilot #11959: 137: error: couldn't find python module jinja2

goonbunny, 2019-05-04:
Hello, 

I am trying to install PX4 developer with Qt creator and have followed the documentation on the dev site exactly. No issues until I try to run the cmake file. The error I get is  

137: error: couldn't find python module jinja2

I know tried all the install jinja2 stuff and they all say its already installed. Any idea why its not getting found? 

Thank you

dagar, 2019-05-05:
It's likely a python problem. Either confusion between python2 & python3 or things like Anaconda.

From a clean state (`make clean`) can you try to `make` again, and share the full output?

goonbunny, 2019-05-05:
Ok, thank you for the help. Here's the code output showing I already have Jinja and `make clean ` on the Firmware folder that has the make file (did I use that command properly?). 

> Colemans-MacBook-Pro:~ colebunny$ easy_install Jinja2
> Searching for Jinja2
> Best match: Jinja2 2.10.1
> Adding Jinja2 2.10.1 to easy-install.pth file
> error: [Errno 13] Permission denied: '/usr/local/lib/python2.7/site-packages/easy-install.pth'
> Colemans-MacBook-Pro:~ colebunny$ pip install Jinja2
> DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
> Requirement already satisfied: Jinja2 in /usr/local/lib/python2.7/site-packages (2.10.1)
> Requirement already satisfied: MarkupSafe>=0.23 in /usr/local/lib/python2.7/site-packages (from Jinja2) (1.1.1)
> Colemans-MacBook-Pro:~ colebunny$ cd Firmware
> Colemans-MacBook-Pro:Firmware colebunny$ make clean

and then here is the output from Qt when I try to run the make file. 

> Running "/usr/local/bin/cmake -E server --pipe=/tmp/cmake-.GpMcbE/socket --experimental" in /private/var/folders/w3/dv2w2ttn6j7_mmw50nf4vr2w0000gn/T/QtCreator-kRojIA/qtc-cmake-YAGpFUMo.
> Starting to parse CMake project, using: "-DCMAKE_CXX_COMPILER:STRING=/usr/bin/clang++", "-DCMAKE_C_COMPILER:STRING=/usr/bin/gcc", "-DCMAKE_PREFIX_PATH:STRING=", "-DQT_QMAKE_EXECUTABLE:STRING=".
> CMake Warning (dev) at CMakeLists.txt:150 (set):
>   implicitly converting 'STRINGS' to 'STRING' type.
> This warning is for project developers.  Use -Wno-dev to suppress it.
> 
> PX4 config file: /Users/colebunny/Firmware/boards/px4/sitl/default.cmake
> PX4 config: px4_sitl_default
> PX4 platform: posix
> PX4 lockstep: enabled
> PX4 version: v1.9.0-rc0-5-g29719894f7
> cmake build type: RelWithDebInfo
> The CXX compiler identification is AppleClang 10.0.1.10010046
> The C compiler identification is AppleClang 10.0.1.10010046
> The ASM compiler identification is AppleClang
> Found assembler: /Applications/Xcode.app/Contents/Developer/usr/bin/gcc
> Check for working CXX compiler: /usr/bin/clang++
> Check for working CXX compiler: /usr/bin/clang++ -- works
> Detecting CXX compiler ABI info
> Detecting CXX compiler ABI info - done
> Detecting CXX compile features
> Detecting CXX compile features - done
> Check for working C compiler: /usr/bin/gcc
> Check for working C compiler: /usr/bin/gcc -- works
> Detecting C compiler ABI info
> Detecting C compiler ABI info - done
> Detecting C compile features
> Detecting C compile features - done
> Found PythonInterp: /usr/bin/python (found version "2.7.10") 
> CMake Error at /usr/local/Cellar/cmake/3.14.3/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
>   couldn't find python module jinja2:
> 
>   		
> 
>   for debian systems try: 		
> 
>   	sudo apt-get install python-jinja2 		
> 
>   or for all other OSs/debian: 		
> 
>   	sudo -H pip install jinja2
> 
>    (missing: PY_JINJA2) 
> Call Stack (most recent call first):
>   /usr/local/Cellar/cmake/3.14.3/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
>   cmake/px4_base.cmake:144 (find_package_handle_standard_args)
>   CMakeLists.txt:351 (px4_find_python_module)
> 
> 
> Configuring incomplete, errors occurred!
> See also "/private/var/folders/w3/dv2w2ttn6j7_mmw50nf4vr2w0000gn/T/QtCreator-kRojIA/qtc-cmake-YAGpFUMo/CMakeFiles/CMakeOutput.log".
> See also "/private/var/folders/w3/dv2w2ttn6j7_mmw50nf4vr2w0000gn/T/QtCreator-kRojIA/qtc-cmake-YAGpFUMo/CMakeFiles/CMakeError.log".
> CMake Project parsing failed.

xdwgood, 2019-05-06:
>  error: [Errno 13] Permission denied: '/usr/local/lib/python2.7/site-packages/easy-install.pth'

maybe try: `sudo easy_install Jinja2`

goonbunny, 2019-05-07:
I tried sudo install but got the same error when I ran c make. Any other ideas? Thank you!!!

julianoes, 2019-05-08:
This sounds like you have python installed multiple times on your machine and you need to figure out which one is used in what environment. There might be Python installed by default, then by brew, Anaconda, etc.

goonbunny, 2019-05-09:
I fixed it!

Step one was to remove python 2.7 and 3 with homebrew
```
brew uninstall python@2
brew uninstall python3
```

Step two was reinstall python 3
`brew install python3`

step three was to install jinja 2 using sudo
`sudo python -m pip install jinja2`

closed 2019-05-09 by goonbunny

Kenshin-T, 2020-10-19:
I fixed it with 
`sudo apt-get install python-jinja2`