PX4/PX4-Autopilot #11960: MC: Undetermined YAW in Loiter

tops4u, 2019-05-04:
**Describe the bug**
When switching to Loiter from MIssion when Copter is close to a Waypoint (i.e. when reaching the last Waypoint and Mission ended) then when switching to Loiter, the Copter will turn YAW always toward Waypoint and then move. So it will always turn YAW randomly. 

**To Reproduce**
Switch to Loiter very close to a Waypoint or at the end of a Mission

**Expected behavior**
Copter tries to keep position without turning around YAW randomly. 

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=e2223cf3-53cc-4779-aaed-c3b3c790b5b0

**Drone (please complete the following information):**
Generic Hexa with FMUv5 (Pixhack v5)

tops4u, 2019-05-08:
To Illustrate what is happening while in Loiter:

[image]

LorenzMeier, 2019-05-08:
Can you describe what you are seeing? I see that it changes heading, but then the change is mostly due to wrapping at 180 degrees.

tops4u, 2019-05-09:
Yes it is changing Heading towards the waypoint I assume. Check the Setpoint which moves. The Cooter actually started
Moving in Yaw unexplainably to me in this moment. Wrapping is not the Problem. I have to check (currently On Mobile) but most probably this can be seen in the Compass Data as well.

bresch, 2019-05-09:
@tops4u Thanks for reporting that. I see that you are using the MPC_YAW_MODE 3 (along trajectory). I checked the log and it looks that the threshold is too small, I'll have to increase it.

For more context, the issue is the following: in that mode, the yaw setpoint is created using the orientation of the velocity setpoint vector. When this vector is small, a small change in X or Y can lead to large yaw variation. There is a guard that locks the yaw setpoint when that vector is small; apparently the value used to define "small" is not large enough.

tops4u, 2019-05-09:
Thanks @bresch for the clarification. 
Maybe I was using it wrong or my Aircraft was not able to track the Position close enough?! I just seemed wrong, that suddendly it started to change yaw in Loiter. Never the less, would one not expect that yaw would be always fixed in loiter independent of the actual Position deviation? Except maybe when commanded by the pilot.

My goal is using a survey but maintaining a fixed YAW Angle during the Flight (so the Heading will be the same i.e. to the East) independent of the flight direction. I know this was possible, but I have new FMU HW and tried to adapt the Parameters. Also in the above Mission I did not set a YAW Heading for the waypoints.

bresch, 2019-05-09:
@tops4u You're using it correctly, the yaw is generated from the velocity setpoint so the tracking is not important, nothing is computed by using the state of the vehicle. I'll fix the issue asap.

tops4u, 2019-05-09:
@bresch May I ask you something completely off topic? I wanted to perform a flight in Survey as explained above, but I can not fix YAW to the heading of the first Survey Leg. Do you know if this is even possible?

BazookaJoe1900, 2019-05-09:
I am not sure if its the same issue or not, but I have seen strange thing in the simulator related to yawing.
the drone does very strange yawing if its get new position setpoint while it still flying to the prev one. its look like either overshoot, or trying to go back to the position where it got re-position.
How to repruduce:
* start iris simulation, default parameters
* takeoff and fly north
* during the flight set new setiont in the east, you will see the drone rotate right, all the way to the south, instead of rotating to the east to the next way point. only afther doing some U-turn its goes to east

attached log to show the issue
[log](https://logs.px4.io/plot_app?log=02c52764-67cf-4c3d-83dd-7d0bd91f13d4)

bresch, 2019-05-10:
> May I ask you something completely off topic? I wanted to perform a flight in Survey as explained above, but I can not fix YAW to the heading of the first Survey Leg. Do you know if this is even possible?

@tops4u I never tried that, sorry. Please create a new post on the [forum](http://discuss.px4.io) and share it in Slack.

@BazookaJoe1900 I'm aware of that behavior and it's not related to the yaw issue. When using goto during flight, the navigator creates a new waypoint where the drone is right now, this is why it's coming back. There is the same problem for RTL. I'll try to find a nicer solution but it might take a bit of time, this is not "critical".

closed 2019-05-30 by LorenzMeier