PX4/PX4-Autopilot #11961: Logfile missing Part 1.9rc

tops4u, 2019-05-04:
**Describe the bug**
Logfile is missing a first Leg. Flight was in Mission Mode consisting of 4 Waypoints, but only last Leg was included in the Log File. SD Card Used is a SanDisk Ultra 32GB (as recommended). Log is HighSpeed Logging of IMU Data.

**To Reproduce**
N/A

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=e2223cf3-53cc-4779-aaed-c3b3c790b5b0

**Drone (please complete the following information):**
Generic Hexacopter
Pixhack v5

bkueng, 2019-05-06:
That is concerning. The log file looks intact, and there are no dropouts recorded in the beginning. Rather it looks like the logger did not log initially.

Do you have the mission log file for that flight?
Did you see this happening before?

bkueng, 2019-05-07:
> On the SD Card?

yes

tops4u, 2019-05-08:
@bkueng Here is the Log
[10_49_20.ulg.zip](https://github.com/PX4/Firmware/files/3158630/10_49_20.ulg.zip)

tops4u, 2019-05-09:
I have a second Log! 

This time I started the Copter in Stabilized, gained Altitude until around 6m then I switched to Position Control Mode. In this moment the Copter started tumbling down! I just managed to switch to Stabilized and regain control about 1m over Ground. I could see the Cube LED flashing rapidly green. I then got it to around 8m Altitude moved it over a field (instead of the concrete where I started). I checked the LIDAR Ground Distance in QGC which indicated 8.62m (I remember this clearly). As it seemed stable at this Moment I started the Mission which completed as expected. Then I flew back home in Stabilized (as I did not trust it completely) and landed. 

Now I wanted to check what happened just after the Start. I'm rather amazed that again most of the Logfile ist Missing on the SDCard as well as in QGC. I could also clearly remember seeing the Warnings about RTL (for an unknown Reason) in QGC. Still the Data is not logged in QGC either! 

SD Card Log:
https://review.px4.io/plot_app?log=cb74137e-a5aa-4482-bf45-59fb88cb64ce

QGC Log (retrieved via iTunes):
https://review.px4.io/plot_app?log=b45ff216-2778-4fe0-987f-37969dde5779

Mission Log:
https://logs.px4.io/plot_app?log=acc264ab-9cd8-43f1-a003-858a8561dcf8

My Setup: 
Pixhack v5 powered from a ACSP4 Board. Drotek uBlox 8 GPS with Compass powered from the Pixhawk. Lightware LW20 powered from the Pixhawk. CUAV WIFI (Powered over the Pixhack). Futaba SBus Receiver powered from the Pixhawk.

Do I draw too much Power from the FMU? I have not seen any Power Alarms for the 5v Bus.

bkueng, 2019-05-09:
I have not yet looked at the second log, but the logging loss is explained as following, and is completely consistent with the log data: log streaming was activated via the CUAV WiFi, and due to unreliable/limited bandwidth, transferring the log header took a long time, causing data to be missed.

tops4u, 2019-05-09:
@bkueng Thanks for your help. I'll switch off the automatic Logging on QGC and follow up the Problems above in a new issue if they reappear.

closed 2019-05-09 by tops4u

hamishwillee, 2019-05-09:
@tops4u Possibly not helpful, but the docs https://docs.px4.io/master/en/getting_started/flight_reporting.html suggest downloading log from vehicle itself for flight reporting. 

Apparently you need a really good connection for reliable telemetry-based logging. When I asked I was told wifi "_might_ be good enough"

[image]

tops4u, 2019-05-10:
@hamishwillee Maybe it would be a good idea if we add a small Note that this may cause a delay in Logging or that it should be switched off if Delayed Logging occurs?

hamishwillee, 2019-05-10:
@tops4u Hmmm. I guess we could update the warning in https://docs.qgroundcontrol.com/en/SettingsView/MAVLink.html#logging

[image]

To:

> **Warning** MAVLink 2 Logging requires a *constant high-rate* MAVLink connection to the vehicle.
  If this is not available then parts of the log may be omitted.
>
>  A basic setup using an RC telemetry link will not be sufficient! A WiFi network may work if the bandwidth can reliably be kept at a high level. 

What do you think @bkueng - is this change an improvement? Note that
- If someone did not read the original warning they might not read this new one
- I have assumed data at any point can be lost, not just the beginning.

tops4u, 2019-05-10:
I think the important point here is that the User (aka me  :-) ) may not be aware that also the SD Card Log will be truncated when the GCS Log has transmission Problems. I assumed they are completely separated.

hamishwillee, 2019-05-10:
Ah, then I am completely confused. I though you were using MAVLink 2 logging, which is streamed ulog,. 
WHen I now think you're saying that you were logging on the SD card on the vehicle, and that this was broken by GCS log transmission problems? I don't get that at all?

tops4u, 2019-05-10:
I was doing both. Hi Frequency Logging to the SDCard and I also had the GCS Logging switched to Automatic. 

Now maybe the HiFrequency Logging (IMU) also applies to the Log which is send to GCS so this would explain the bottleneck of the Wifi... 

As far as I understood Beat, all Logging only starts when all Loggers have successfully received the Log Header, which may be delayed due to congestion on the GCS Downlink.

@bkueng correct?

bkueng, 2019-05-14:
Correct, this is a design decision that considerably reduces resource usage. It only affects the log header, and the case we have here is an extreme manifestation of that.
Generally log streaming is most helpful to stream to a companion, with known available bandwidth.

hamishwillee, 2019-05-15:
@tops4u @bkueng 

Our docs currently say "PX4 logs detailed aircraft performance data once the system has been armed until it is disarmed.". There is nothing that says or indicates that something needs to be downloaded or even that a GCS needs to be connected for this.

So I don't get ....

> As far as I understood Beat, all Logging only starts when all Loggers have successfully received the Log Header, which may be delayed due to congestion on the GCS Downlink.

Or even what the log header is.

Can I have the even bigger dummies guide?

hamishwillee, 2019-05-15:
Or alternatively, add the note where you think it should go in the docs, and I'll tell you if it makes sense as a reader.

bkueng, 2019-05-15:
> Or even what the log header is.

So this is only relevant if (QGC) log streaming is enabled. I don't think the docs need updating.