PX4/PX4-Autopilot #11967: ATT_POS_MOCAP can not work with EKF without ROS

machens, 2019-05-06:
ATT_POS_MOCAP can not work with EKF2

We tested LPE with the below code by sending ATT_POS_MOCAP and it works fine. Hardware: PixRacer R14, Firmware: 1.8.2, MoCap: Optitrack.

[https://github.com/machens/Optitrack-SampleClient](https://github.com/machens/Optitrack-SampleClient)

But if we switch to EKF2 with same enviorment, LOCAL_POSITION_NED didnot appear in Mavlink Inspector and system reject Position Control mode. We set the below parameter.

EKF2_AID_MASK = 24
EKF2_HGT_MODE = Vision

[In Link](https://dev.px4.io/en/ros/external_position_estimation.html), we find this feature has only been tested to work with the LPE estimator.

[In Link](https://dev.px4.io/en/tutorials/motion-capture-vicon-optitrack.html), it is related ROS, but we dont use ROS.

How to make it work with EKF2? 

Thanks!

TSC21, 2019-05-06:
It's documented that this message data is not propagated in uORB to be consumed by EKF2. If you want to use mocap data, just use msgs like ODOMETRY or VISION_POSITION_ESTIMATE.

closed 2019-05-06 by TSC21

hamishwillee, 2019-05-06:
The links are the v1.8.2 docs. We recently updated these in the master branch - see https://dev.px4.io/master/en/ros/external_position_estimation.html (note, the docs still apply to 1.8.2, it's just that the 1.8.2 branch was created before the update).

So see https://dev.px4.io/master/en/ros/external_position_estimation.html