PX4/PX4-Autopilot #11970: Support reset_counter of VISION_POSITION_ESTIMATE message

michael-p, 2019-05-06:
**Describe problem solved by the proposed feature**
Mavlink message `VISION_POSITION_ESTIMATE` has a field `reset_counter` ([Mavlink documentation](https://mavlink.io/en/messages/common.html#VISION_POSITION_ESTIMATE)) which is very useful if the position estimates are generated by a SLAM system (that is, not just visual odometry) but it looks like this is field currently ignored in the PX4 firmware.

**Describe your preferred solution**
Pass `reset_counter` through to the ekf and make sure that large position jumps from the vision system do not cause any problems there.

TSC21, 2019-05-06:
This is a quite new field - it's not that is ignored, it's just not handled yet.

TSC21, 2019-08-04:
Still relevant.

james-modalai, 2020-03-18:
Bringing this back from the dead. We are trying to get relocalization via loop-closure and fiducial markers working and sure enough I find PX4 ignores this field after getting everything else working. Any pointers would be appreciated as I evaluate what it would take to implement this.

PLMR-FR, 2020-08-09:
Could be very usefull if it was supported for SLAM system.
Is it in the roadmap ?

michael-p, 2021-04-15:
@james-modalai Did you get to any conclusions regarding this issues and possible implementation? I just bumped into this issue in a different project again... :)

james-modalai, 2021-04-15:
We've just been giving px4 continuous inputs and keeping track of offsets
offboard. This will be a problem when swapping back and forth from GPS to
VIO though.

dagar, 2021-04-15:
If you can start populating the `VISION_POSITION_ESTIMATE` reset_counter we can look at an initial implementation PX4 side. We've already done the work in the controllers to handle the delta in different estimates (position, velocity, attitude), it would be great to handle this cleanly across the entire system.

james-modalai, 2021-04-16:
Thank's Daniel! we've been populating the reset counter since we first got
VIO going with PX4, I've just avoided doing any resets or relocalization
jumps mid-flight. I'll give it a go with px4 master branch soon.

bresch, 2023-07-28:
The reset counter is now supported; the EKF resets its states when received: https://github.com/PX4/PX4-Autopilot/blob/444e5d2d4abbb0adaabf1ad3cddaa2f9d2f9ee2b/src/modules/ekf2/EKF/ev_control.cpp#L50

closed 2023-07-28 by bresch