PX4/PX4-Autopilot #11975: Fixedwing: implement IAS output scaling

bresch, 2019-05-06:
In the devguide ([here](https://dev.px4.io/master/en/flight_stack/controller_diagrams.html#rate-ff-scaling)) I demonstrated that the feedforward (FF) part of the rate controller has to be scaled linearly with the indicated airspeed (IAS). At the moment, both [PI quadratic scaling](https://dev.px4.io/master/en/flight_stack/controller_diagrams.html#static-torque-pi-scaling) and FF linear scaling are done with true airspeed (TAS):
https://github.com/PX4/ecl/blob/c4492b17c1d7bb9b74011efa9890452b091421e8/attitude_fw/ecl_roll_controller.cpp#L111-L113

Since IAS depends on air density, this makes a real difference when flying at high altitudes, cold or warm places.

Antiheavy, 2019-08-07:
Agree, it makes sense to use TAS here.  Although it would be nice to have better ways to estimate outside air temperature (OAT) in PX4.  For example, our airspeed sensor heats up significantly in the cavity it is mounted in causing PX4 TAS estimate to be wrong.  I've seen this handled in other systems by allowing a OAT offset parameter or also allowing different sources of OAT measurement, not just from the Diff Press sensor.

supergra, 2020-07-02:
@Antiheavy Are you sure that the temperature is affecting your TAS measurement? Last time I checked the code, it seemed that PX4 just assumes the OAT is 20C and doesn't trust the temperature at all.

I agree with the need to compute TAS more accurately, but I don't think I've seen fluctuations in TAS even though the temperature measurements were wildly wrong.