PX4/PX4-Autopilot #11980: 1.9.0 RC0 does not start properly on Pixhawk2 Cube

FlavioTonelli, 2019-05-07:
When I start a pixhawk2 cube board running PX4 1.9.0 RC0 the boot sound stop suddenly.
I'm wondering if this behavior is an indicator of a bug in the last release.
I can reproduce the same behavior flashing on the board the last master(commit 6d8d8e).

Here you can find two videos:
One performing a boot running 1.8.2: [1.8.2 Boot](https://drive.google.com/open?id=1AeWsxHVQ8T5hBV7ThXFIavD3wYDO7taW)
The other one running 1.9.0RC0:  [1.9.0 Boot](https://drive.google.com/open?id=1hFA92UxPoHvzxpqBpe5PonBhXOko-vVY)

LorenzMeier, 2019-05-07:
@bkueng Could you have a look if Daniel doesn't beat you to it?

bkueng, 2019-05-08:
I can reproduce this. The board boots up fine, but the startup sound stops midway.

bkueng, 2019-05-08:
Fixed in https://github.com/PX4/Firmware/pull/11988.