PX4/PX4-Autopilot #11986: unable to start  mission : vehicle not  ready

sampathkumarch, 2019-05-08:
https://discuss.px4.io/t/unable-to-start-mission-vehicle-not-ready/10896

closed 2019-05-08 by julianoes

sampathkumarch, 2019-05-09:
> Sir I have these components
>
> 1.pixhawk
> 2.gps module
> 3.power management board
> 4.lipo battery
> 5. One ESC
> 6.one brush less motor
> 7. Q ground control software
>
>   I don't have these comments
>
> 1.s.bus transmitter and receiver
> 2.telemetry system
> 3. Air speed sensor
>
> Wiring connection:
> throttle pin is given IO port 4 pin
> Right elevator is given IO port 2 pin
> Left elevator is given IO port 1pin
>
>
> Frame : FIXED FLYING WING
>
>
> I followed these steps
>
> 1.upload firmware
> 2.frame selected ( generic flying wing)
> 3.calibrate all sensors but air speed sensor is not calibrate so using
> circuit breaker disabled this
> 4.power calibration completed
> 5 safety calibrated
>
> I do not have a receiver and transmitter so I can run in autopilot so
> COM_RC_IN_MODE was desiabled.then after you say it NAV_RC_ACT is set 0 .
>  Am using USB cable for connectivity computer to pixhawk so that one also
> disabled using circuit breaker.
> Then after you say  battery checks also disabled by using circuit breaker.
> Then after I given a mission plane and upload it then show the one message
> is start the mission.then after I start the mission getting one more
> message is unable to start  mission .vehicle not ready .please resolve my
> problem sir. please help me sir

sir please help me . why not given to me reply tell me sir

hamishwillee, 2019-05-09:
@sampathkumarch 

> why not given to me reply tell me sir

The issue was closed because this is a place for bug reports and feature requests - is not a place for you to post support requests. You have correctly created a discussion board issue. Additional information and questions should go in that issue.

sampathkumarch, 2019-05-10:
Thank you for your response, but where is the discussion this issue iam
post for it

hamishwillee, 2019-05-10:
Pardon? In _your_ top post https://github.com/PX4/Firmware/issues/11986#issue-441587312 -  see https://discuss.px4.io/t/unable-to-start-mission-vehicle-not-ready/10896

hamishwillee, 2019-05-10:
If that isn't your post, then please create a new one. We can't help you in the defect tracker.

sampathkumarch, 2019-05-10:
Ok sir  .what do I do now?

sampathkumarch, 2019-05-10:
Ok sir, what I do now

hamishwillee, 2019-05-10:
Create a post in https://discuss.px4.io with all the details of your setup - vehicle, flight controller, autopilot, autopilot version, ground station, what you have tried. Everything you can think of that might be relevant to the problem.

hamishwillee, 2019-05-10:
The information you have put in this issue looks good, but it can't be posted here, because this is for bugs, and at the moment all we know is that you have a problem. That might be due to your setup or something else.