PX4/PX4-Autopilot #11989: TCA62724 and TCA62724 will not light up on some PX4_I2C_BUS_EXPANSIONs

bys1123, 2019-05-08:
**Describe the bug**
TCA62724 only can light up on Pixhawk4 GPS module port(bus 1). TCA62724 can light up on GPS module port(bus 1) and UART I2C B (bus 2), but they both can't light up on Pixhawk4 I2C A(bus 3).

Tested on 1.9.0-rc1

**Expected behavior**
TCA62724 and NCP5623C should light on every buses.

**Additional context**
Some drone show manufactures are using TCA62724 I2C protocol to light their owm lights in china. One of them is changing their autopilot to Pixhawk4. On Pixhawk4, that means they hard to use the old way to light their LED.

I tried to change the driver or boot script, but this two driver looks the same, don't know why they behave different, so I have no idea to change the code.

bys1123, 2019-05-17:
@julianoes Do you have time open this driver for all I2C buses?

julianoes, 2019-05-20:
I'm not sure I know what to do here but I'm pinging @davids5 and @bkueng hoping that they know more.

davids5, 2019-05-20:
@bys1123 

This is the bus selection https://github.com/PX4/Firmware/blob/master/src/drivers/lights/rgbled_ncp5623c/rgbled_ncp5623c.cpp#L372-L390
```

based on these

Simple Solution is to add 'B' or 'R' as an option and pass the the real bus number to the code.
/* I2C busses */

#define PX4_I2C_BUS_EXPANSION       1
#define PX4_I2C_BUS_EXPANSION1      2
#define PX4_I2C_BUS_ONBOARD         3
#define PX4_I2C_BUS_EXPANSION2      4
#define PX4_I2C_BUS_LED             PX4_I2C_BUS_EXPANSION
```

julianoes, 2019-08-19:
@bys1123 has this been resolved?

closed 2019-08-26 by bys1123