PX4/PX4-Autopilot #11991: failure in sitl with mavros connection in local.

sizsJEon, 2019-05-09:
I'm currently trying to control my virtual drone (which run by "make px4_sitl_default gazebo" )

I'm succeed on this, so I could see drone takes command  like "commander takeoff"

but I'm failing with mavros's px4 controller.(which run by roslaunch mavros px4.launch  fcu_url:="udp://:14540@127.0.0.1:14556")
and changing port and my ip also failed.
(like roslaunch mavros px4.launch fcu_url:="udp://:14550@192.168.10.106:14557"

could you help me for run mavros px4 node?

below is the log for each node.

**1) root@jeon-ml:/home/jeon-ml/ros_jeon/src/Firmware# make px4_sitl_default gazebo**
[  0%] Built target lockstep_scheduler
[  0%] Built target mixer_gen
[  2%] Built target df_driver_framework
[  2%] Built target uorb_headers
[  2%] Built target ver_gen
[  3%] Built target mixer_gen_6dof
[  3%] Built target git_ecl
[  3%] Built target git_gps_devices
[  3%] Built target git_mavlink_v2
[  3%] Built target logs_symlink
[  4%] Built target git_gazebo
[  4%] Built target tinybson
[  4%] Built target perf
[  6%] Built target work_queue
[  6%] Built target ecl_airdata
[  7%] Built target ecl_attitude_fw
[  7%] Built target ecl_geo
[ 22%] Built target uorb_msgs
[ 22%] Built target ecl_geo_lookup
[ 22%] Built target ecl_tecs
[ 22%] Built target ecl_validation
[ 23%] Built target mixer
[ 24%] Built target rc
[ 24%] No forceconfigure step for 'sitl_gazebo'
[ 24%] Built target generate_topic_listener
[ 24%] Performing configure step for 'sitl_gazebo'
[ 25%] Built target parameters
[ 26%] Built target ecl_l1
-- install-prefix: /usr/local
-- Using C++14 compiler
[ 27%] Built target px4_layer
[ 28%] Built target px4_daemon
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   thread
--   filesystem
--   chrono
--   date_time
--   atomic
[ 29%] Built target modules__uORB
-- Boost version: 1.58.0
[ 30%] Built target px4_platform
[ 31%] Built target systemlib
[ 32%] Built target cdev
-- Building klt_feature_tracker without catkin
-- Building OpticalFlow with OpenCV
[ 32%] Built target airspeed
[ 34%] Built target lib__cdev__test__cdev_test
[ 34%] Built target battery
[ 34%] Built target CollisionPrevention
[ 34%] Built target circuit_breaker
[ 36%] Built target controllib
[ 36%] Built target conversion
[ 36%] Built target drivers__device
[ 36%] Built target lib__controllib__controllib_test
[ 37%] Built target mathlib
[... 382 lines cut ...]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to df30f2de-722f-11e9-8d45-708bcd505716
process[rosout-1]: started with pid [32756]
started core service [/rosout]
process[mavros-2]: started with pid [306]
[ INFO] [1557388555.163134798]: FCU URL: udp://:14540@127.0.0.1:14556
[ INFO] [1557388555.164120781]: udp0: Bind address: 0.0.0.0:14540
[ INFO] [1557388555.164149316]: udp0: Remote address: 127.0.0.1:14556
[ INFO] [1557388555.164240772]: GCS bridge disabled
[ INFO] [1557388555.164600825]: udp0: Remote address: 127.0.0.1:14580
mavros_node: /usr/include/boost/smart_ptr/shared_ptr.hpp:641: typename boost::detail::sp_dereference<T>::type boost::shared_ptr<T>::operator*() const [with T = boost::signals2::detail::signal_impl<void(const __mavlink_message*, unsigned char, unsigned char), boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function<void(const __mavlink_message*, unsigned char, unsigned char)>, boost::function<void(const boost::signals2::connection&, const __mavlink_message*, unsigned char, unsigned char)>, boost::signals2::mutex>; typename boost::detail::sp_dereference<T>::type = boost::signals2::detail::signal_impl<void(const __mavlink_message*, unsigned char, unsigned char), boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function<void(const __mavlink_message*, unsigned char, unsigned char)>, boost::function<void(const boost::signals2::connection&, const __mavlink_message*, unsigned char, unsigned char)>, boost::signals2::mutex>&]: Assertion `px != 0' failed.
================================================================================REQUIRED process [mavros-2] has died!
process has died [pid 306, exit code -6, cmd /home/jeon-ml/ros_jeon/devel/lib/mavros/mavros_node __name:=mavros __log:=/root/.ros/log/df30f2de-722f-11e9-8d45-708bcd505716/mavros-2.log].
log file: /root/.ros/log/df30f2de-722f-11e9-8d45-708bcd505716/mavros-2*.log
Initiating shutdown!
================================================================================
[mavros-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

#above is end of log

julianoes, 2019-05-13:
@sizsJEon can you try to add your IP as described in https://dev.px4.io/master/en/simulation/#modify-configuration-for-external-broadcasting.

Basically you set `mavlink start ... -t IP` here: https://github.com/PX4/Firmware/blob/77097b6adc70afbe7e5d8ff9797ed3413e96dbf6/ROMFS/px4fmu_common/init.d-posix/rcS#L241

sizsJEon, 2019-05-13:
> @sizsJEon can you try to add your IP as described in https://dev.px4.io/master/en/simulation/#modify-configuration-for-external-broadcasting.
>
> Basically you set `mavlink start ... -t IP` here: [...]

I tried it and failed again.
and for now, I'm going to start over againg from formatting the drone and re-install ubuntu.

when it happens again, I'll open another thread.

thx for reply julianoes

closed 2019-05-13 by sizsJEon