PX4/PX4-Autopilot #11992: Support HobbyKing car ESC

ItsTimmy, 2019-05-09:
**Describe problem solved by the proposed feature**
Support HobbyKing ESC

**Describe your preferred solution**
A clear and concise description of what you want to happen.

**Describe possible alternatives**
A clear and concise description of alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots for the feature request here.

TODO: Fill out this issue

ItsTimmy, 2019-05-09:
@KatrinMoritz

KatrinMoritz, 2019-05-10:
I did some changes in the mixer file `ugv_generic.main.mix` and now the Rover can drive forward, reverse and it's possible to steer.

```
Output 3
---------------------------------------
M: 1
S: 0 3      0  20000 -10000 -10000  10000
```

But Arming is just possible when the left stick is in the lower position. And in this case the car starts driving reverse with full speed.

In the documentation on https://dev.px4.io/en/concept/mixing.html it is mentioned that throttle can be scaled from 0..1 or  -1...1. I think it is scaled from 0..1 and I had not found where to change to -1..1.

julianoes, 2019-05-13:
> In the documentation on https://dev.px4.io/en/concept/mixing.html it is mentioned that throttle can be scaled from 0..1 or -1...1. I think it is scaled from 0..1 and I had not found where to change to -1..1.

Exactly, I think we should figure out how we can enable it from -1..1 for rovers.

julianoes, 2019-05-16:
Discussed in the call:
- Should we use throttle from -1 to 1 and change this everywhere?
- Should we use a separate channel for switching to reverse.
- We could disable arming/disarming using the stick.

- We need to keep the ESC calibration in mind. 

@dagar will do an end to end pass to check this.

- We potentially need to stop at 0 before going to reverse. 
- What about breaking? It should be automatic as it happens when you reduce throttle. It's depending on the ESC.

jarivanewijk, 2019-05-17:
> I did some changes in the mixer file `ugv_generic.main.mix` and now the Rover can drive forward, reverse and it's possible to steer.
>
> ```
> Output 3
> ---------------------------------------
> M: 1
> S: 0 [...]

I have an arming switch assigned on the controller, but even then it will only arm when the throttle stick is all the way down (full reverse). I don't get any warnings though, it just refuses to arm when the throttle stick is not in its lowest position.

Also, once armed, disarming is only possible after using the kill switch and with the throttle stick all the way down. It might still think it's "in flight" or something? Again, no warnings.

julianoes, 2019-08-05:
@dagar do you know the status regarding the throttle things?

jarivanewijk, 2019-08-14:
> @dagar do you know the status regarding the throttle things?

I would also like to know the status. I will try to join the next rover call, but I probably won't make it. We have a small HobbyKing car and a RC boat here with this kind of ESCs. I would like to help with this, but I'm not sure how.

igalloway, 2019-09-26:
I can see merit in using a separate channel to switch into reverse. It certainly would simplify the arming problem.

tylercorn19, 2019-10-16:
I agree with Iain, it could be integrated into a separate channel so that it would be similar to a real car in which you have to switch gears to drive forward/reverse.

jarivanewijk, 2020-01-16:
I have not kept up with rover developments in the last few months, but I think a solution for reversing driving direction has not been implemented yet? I think there is still interest in such a feature.

@dagar @julianoes @ItsTimmy

ItsTimmy, 2020-01-16:
I agree that being able to reverse is an important feature. But I am not actively working on the rover right now, so hopefully someone else can work on it.

brentsony, 2020-06-03:
What parts of the code would need to be modified to support REVerse? 

 How can the arming feature be modified in a custom Airframe so that it is controlled by an RC switch, rather than the current method of min thrust + max yaw?

jbeyerstedt, 2020-06-11:
@brentsony But even if you have a separate arming switch, arming will only be possible with the throttle stick down. So if you adapt the mixer file to scale the RC input properly for driving backwards, you are forced to arm with full throttle backwards, which is obviously not useable.
(At least this was the state, when I looked at this topic the last time.)

But I can confirm, that driving backwards in offboard mode works very well with the standard mixer setup. You just can not drive backwards with the remote control.

brentsony, 2020-06-19:
@Jaeyoung-Lim that works well. Some RCs have a button, which can be assigned to ARM.

@jbeyerstedt The throttle can be assigned to the Right stick using a custom airframe, so it auto-centers. Then a workaround is arming can be done with the ESC off, followed by turning the ESC on.

But PX4 support for rovers seems to be alpha quality at best. Reverse is required in manual mode at a minimum in order to get out of a tight spot.

Is full Land/Water Rover/Boat support on the roadmap?

Jaeyoung-Lim, 2020-06-19:
@brentsony This is definitely part of the roadmap. Would be awesome if you have a use case and help improve rover / boat in PX4

berndpfrommer, 2021-04-13:
Having a way to give negative throttle under RC control is definitely desirable, not least because it is needed to decelerate. On hobby RC cars this is usually done by having some part of the throttle travel reserved for backwards, the rest is for forwards. That's the most intuitive from a user control.

Jaeyoung-Lim, 2021-04-13:
@berndpfrommer Fully agreed :)

LorenzMeier, 2021-04-13:
We actually solved this months ago. If you keep your throttle center during calibration in QGC in the step where it asks you to go all the way to the bottom, you have manual reverse support.

closed 2021-04-13 by LorenzMeier

berndpfrommer, 2021-04-21:
Tried this procedure today:
1) When prompted to take the throttle stick all the way down, leaving it at center stalled the calibration procedure, I *had* to move it down. Fine, so I only moved the throttle to about 1/4 way, leaving some travel left for reverse. That way I could continue with the rest of the calibration.
2) When testing, now indeed the ESC already goes to zero when the throttle is at about 1/4, but no reverse. I checked the output of servo_raw in QGC: between throttle 0 and throttle 1/4, the output to the ESC stays at 1500. Note that I'm using generic ground vehicle as airframe.

Prasanna-icefire, 2021-07-09:
> Tried this procedure today:
>
> 1. When prompted to take the throttle stick all the way down, leaving it at center stalled the calibration procedure, I _had_ to move it down. Fine, so I only moved the [...]

Yup.......Did not reverse....used sempede 2wd

palahmd, 2021-11-19:
Thank you @LorenzMeier, it worked out for me. I was searching for a solution for a while and ended up with the same solution other users have stated above to enable reverse throttle. For the "Arming denied: high throttle" warning I changed the threshold in the comander.cpp file to <0.4 && >0.6 instead of >0.0 so that a centered stick will no more be recognized as a high throttle which is not optimal.

I wish your solution with the RC calibration would have been described in the PX4 docs as well.

berndpfrommer, 2021-11-30:
@palahmd Could you please comment on what exactly your procedure was to get this working? I'm not sure what "same solution as other users have stated above" is exactly referring to. Thanks!

palahmd, 2021-11-30:
@berndpfrommer I have got it working with a similar solution like KatrinMoritz:

> I did some changes in the mixer file `ugv_generic.main.mix` and now the Rover can drive forward, reverse and it's possible to steer.
>
> ```
> Output 3
> ---------------------------------------
> M: 1
> S: 0 [...]

However, this is not optimal, since the left stick must be in the lower position (which equals a throttle signal of 0.0) when arming, which will produce reverse full throttle instantly after the vehicle is armed. If you center the stick, you will get a warning "arming denied: high throttle". You can change this warning/arming condition to a range outside of [0.4, 0.6] instead of >0.0 in the Comander.cpp file, which is also not optimal. You can then arm with a centered left stick (which equals a throttle signal of 0.5) without getting this warning. 

By calibrating the RC with the instructions of LorenzMeier, reverse thrust is enabled and a centered left stick equals 0.0 (and a lowered left stick equals -1.0):

 

> We actually solved this months ago. If you keep your throttle center during calibration in QGC in the step where it asks you to go all the way to the bottom, you have manual reverse support.

berndpfrommer, 2021-12-01:
@palahmd Thanks much for filling in the details. Will try that once I get the car back up and running!

jakobjackson2016, 2022-02-21:
> For the "Arming denied: high throttle" warning I changed the threshold in the comander.cpp file to <0.4 && >0.6 instead of >0.0 so that a centered stick will no more be recognized as a high throttle which is not optimal.

@palahmd and @berndpfrommer Could you please share the exact code line #(s) where you made this change in the comander.cpp file if possible? Thanks!

zouzhe1, 2022-08-13:
@julianoes @palahmd  @jbeyerstedt @Jaeyoung-Lim 
I also want to use the throttle back function. There are 2 situations in total. First, the throttle stick is brought back to the middle function. Push the throttle up to go forward, and pull down the throttle to go backward. Second, the throttle stick does not have the function of returning to the center, and a remote control toggle is required to switch forward and backward. Is it possible to implement the back function now? I didn't understand the mix you were talking about.
thanks

zdeno-sekerak, 2026-08-28:
Has this issue been resolved somehow? I am still having problems with setting up reverse on PX4 1.16 using CAN to control a VESC. Instead of reversing, the car just drives forward at the same speed, no matter the negative stick position.

zdeno-sekerak, 2026-08-28:
That is me