PX4/PX4-Autopilot #11993: Add PixyCam2 driver for Rover

igalloway, 2019-05-09:
For general questions please use [PX4 Discuss](http://discuss.px4.io/) or [Slack](http://slack.px4.io/).

**Describe problem solved by the proposed feature**
NXP will use the PixyCam as for a simple car configuration which is able to stay inside the lines on either side of a race track. This is development that will be used for student competitions. 

**Describe your preferred solution**
Driver in PX4 that provides access to PixyCam pre-programmed functions

**Describe possible alternatives**
Alternative would be for PixyCam to use the SDK to command the FMU. But this is not preferred.

**Additional context**
Leo Mustafa (NXP) has started a driver and communications between the Pixy and PX4, so there is a solid starting point

jarivanewijk, 2019-08-07:
As mentioned, some basic code already exists (not publicly available). We need to figure out if can release it and if it is worth it to bring it upstream.

jpt48, 2019-10-30:
Good morning,
We are currently having a file error issue when we try to compile the code. it may be worth saying that the px4_config.h file it is in the correct directory, any suggestions ?? Thank you
[image]

bkueng, 2019-10-31:
@jpt48 the include paths changed on master: `<px4_config.h>` --> `<px4_platform_common/px4_config.h>`

jpt48, 2019-11-18:
@bkueng Thank you, the errors were cleared. 
We have come across another issues if anyone may have input. The FMU took the updated firmware with Pixy cam code. To test it we are trying to get feedback through QGC mavlink and this its all we get back. Thank you.
[image]

bkueng, 2019-11-19:
What output do you expect? Might be best to open a pull request (even if not finished yet), so we can look at the code.

jarivanewijk, 2020-02-18:
(Note that the PixyCam driver has been available at https://github.com/NXPHoverGames/PixyCam for a while now. Should we again bring up the discussion whether to bring this into PX4?)

julianoes, 2020-03-02:
@jarivanewijk we're always open for pull requests but I'm not sure if you can convince someone to work on this for you.

jarivanewijk, 2020-03-31:
@igalloway @KatrinMoritz - Is this still something we want to merge into PX4 eventually? Otherwise this issue can be closed.

KatrinMoritz, 2020-04-01:
Yes, we want this. And I will do this PR soon.

KatrinMoritz, 2020-04-16:
@jarivanewijk @igalloway @julianoes the issue can be closed. There will be no PR for the Pixycam driver.

closed 2020-07-21 by julianoes