PX4/PX4-Autopilot #12000: Smooth takeoff broken?

MaEtUgR, 2019-05-10:


MaEtUgR, 2019-05-11:
I'm working on a fix: #12001

AuterionWrikeBot, 2019-05-13:
➤ Matthias Grob commented:

It really bugged me and so I had a closer look and tried to find a good solution on Saturday. We need to flight test it today on the F450.

Mathieu Bresciani please take none of the comments personal, it's really only about the minimal working functionality and we can replace the implementation when we have a better solution that cannot negatively impact user experience. For the smooth vel tasks the takeoff is still triggered by jerk and I'd appreciate your feedback.

MaEtUgR, 2019-05-16:
Update: Since #12001 does not satisfy the expectations for the release there's the revision on #12014.

closed 2019-05-22 by LorenzMeier