PX4/PX4-Autopilot #12007: Downloading Log files causes Mag Errors

tops4u, 2019-05-13:
**Describe the bug**
While downloading Logfiles (> 10MB) the Cube started flashing Pink. Now by redoing it I noticed that the Cubes is "Breathing" is "stuttering" while Log Downloads. Sometimes it would timeout the Mags:

[image]

[image]

I do not think this is Mission-Critial, but still worth noting. 

**Drone (please complete the following information):**
CUAV Pixhack v5, PX4 1.9 RC2

julianoes, 2019-05-13:
Mag timeouts are usually an indication of the system being stalled, and not actually a problem of the mag itself. This would mean something in MAVLink FTP is consuming too much CPU or IO and impacting the system overall.

Antiheavy, 2019-05-13:
> Mag timeouts are usually an indication of the system being stalled, and not actually a problem of the mag itself.

We run into this annoying issue quite often.  is there a strategy that could solve it?

dagar, 2019-05-13:
There are a few different cases, but some of them will go away after https://github.com/PX4/Firmware/pull/11571 is merged. Other things running in hpwork (land detector, shutdown logic, etc) will no longer be able to block mag readings.

tops4u, 2019-08-12:
I probably have to retest this, but I AFAIR it still happens in 1.9.x

junwoo091400, 2022-11-22:
@tops4u Could you check again if this still happens now? Although it is super old issue 😆