PX4/PX4-Autopilot #12011: Support Climb speed and Descent speed (CMD_DO_CHANGE_SPEED)

FlavioTonelli, 2019-05-13:
If I get well from the source code, PX4 can right now only support DO_CHANGE_SPEED command when the specify speed type is a ground speed.
In my experience this is quite a big limitation when planning mission with a multicopter vehicle.
In a lot of applications, such that vertical scanning, the climb speed and descent speed are as much significant as the XY one.

julianoes, 2019-05-14:
Fair request. And the MAVLink command already defines how to do that, it's just PX4 that doesn't support it.
https://mavlink.io/en/messages/common.html#MAV_CMD_DO_CHANGE_SPEED

https://github.com/PX4/Firmware/blob/20adce50778f835b1787f17a168e0b1061bbb593/src/modules/navigator/navigator_main.cpp#L469-L487

FlavioTonelli, 2020-05-08:
Hello @julianoes, this is still a desirable feature.
What's the actual state of that? Can I provide any help on Developing/Testing it?